YosysHQ/nextpnr #10: Project mode with CLI

olofk, 2018-08-01:
The project mode seems quite handy for FuseSoC. My idea is to generate a project file with all required properties and then let the user open it in the GUI or use it on the command-line to generate the bitstream with just the project file as input (This is how many of the other tool backends are implemented).

I can't figure out how to use project files on the command-line though. `nextpnr-ice40 --load blinky.proj` just seems to load the file and then exit (also, adding --gui doesn't do anything when --load is used).

mmicko, 2018-08-02:
This was experimental feature added to make a possible all needed parameters to be defined in a file.
It is not doing synthesis part (as other FuseSoC backends) but still would like to hear feedback about possible use cases for this. 
Have done some small changes, since cleanup of parameters actually removed info for gui and asc, so if you now do:
../nextpnr-ice40 --load blinky.proj --asc blinky.asc
that will generate output for you.
Also you can add --gui at the end and get all loaded in gui.
To be honest, I still do not like fact that it depends of options, and we still need to define it better to cover other targets (like ecp5) since it is part of ice40 only now.

Issue is addressed in https://github.com/YosysHQ/nextpnr/pull/11

q3k, 2018-08-04:
@olofk Would you mind verifying that this implementation is adequate for your desired workflow?

olofk, 2018-08-09:
Sorry for the late reply. Turns out that I'm getting `ERROR: Unexpected EOF in JSON file.` when running `nextpnr-ice40 --load blinky.proj` both with and without the `--gui` switch

mmicko, 2018-08-10:
@olofk this could be the case when there is no .json file at all, you would need to generate it first with blinky.sh

olofk, 2018-08-10:
Ah, of course. Sorry. I assumed the unexpected EOF was referring to the project json file. It works as expected now. For the future it would be nice to have target frequency and output file in the project file as well, but this issue can be closed now. Thanks

closed 2018-08-10 by olofk