YosysHQ/nextpnr #3: Running Pack From GUI Results in SIGSEGV on MinGW64 Windows

cr1901, 2018-08-01:
## Environment
```
$ g++ -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\g++.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-7.3.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libexecdir=/mingw64/lib --enable-bootstrap --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,objc,obj-c++,fortran,ada --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-time=yes --enable-libstdcxx-filesystem-ts=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev2, Built by MSYS2 project' --with-bugurl=https://sourceforge.net/projects/msys2 --with-gnu-as --with-gnu-ld
Thread model: posix
gcc version 7.3.0 (Rev2, Built by MSYS2 project)

$ nextpnr-ice40 -V
nextpnr-ice40 -- Next Generation Place and Route (git sha1 2743d0f)
```

## Steps to Duplicate
1. Run `nextpnr-ice40 --gui` from MSYS2/MinGW64 `bash`
2. Design > Open JSON > blinky.json
3. Design > Open PCF > blinky.pcf
4. Design > Pack

## Expected Behavior
No SIGSEGV

## Actual Behavior
Packing starts but soon (~1-2 seconds) results in a "Segmentation Fault" and the application closes.

Backtrace/SIGSEGV source is here (ignore the extra `--asc` option): http://ix.io/1iUS

q3k, 2018-08-01:
Relevant part of stacktract:
```
(gdb) bt
#0  nextpnr_ice40::TreeModel::IdStringList::updateElements (
    this=this@entry=0x156b6ba0, ctx=ctx@entry=0xd5c7fd0, elements=...)
    at ../gui/treemodel.cc:74
#1  0x000000000047d928 in nextpnr_ice40::TreeModel::Model::updateCellsNets (
    this=0x12fdf490, ctx=0xd5c7fd0) at ../gui/treemodel.cc:221
#2  0x0000000000468499 in nextpnr_ice40::DesignWidget::updateTree (
    this=0x12fd7d60) at ../gui/designwidget.cc:245
#3  0x00000000004621e6 in nextpnr_ice40::DesignWidget::qt_static_metacall (
    _a=<optimized out>, _id=<optimized out>,
    _c=QMetaObject::InvokeMetaMethod,
    _o=0x461e60 <nextpnr_ice40::DesignWidget::qt_static_metacall(QObject*, QMetaObject::Call, int, void**)>)
    at generated/gui/ice40/gui_ice40_autogen/EWIEGA46WW/moc_designwidget.cpp:162
#4  nextpnr_ice40::DesignWidget::qt_metacall (this=<optimized out>,
    _c=<optimized out>, _id=<optimized out>, _a=<optimized out>)
    at generated/gui/ice40/gui_ice40_autogen/EWIEGA46WW/moc_designwidget.cpp:239
#5  0x000000000d9e567a in QMetaObject::activate(QObject*, int, int, void**) ()
   from C:\msys64\mingw64\bin\Qt5Core.dll
#6  0x000000000045a754 in nextpnr_ice40::MainWindow::pack_finished (
    this=0x22fa70, status=<optimized out>) at ../gui/ice40/mainwindow.cc:457
#7  0x000000000d9e567a in QMetaObject::activate(QObject*, int, int, void**) ()
   from C:\msys64\mingw64\bin\Qt5Core.dll
#8  0x0000000000460a53 in nextpnr_ice40::TaskManager::pack_finished (
    this=0xffd0208,
    _t1=<error reading variable: Cannot access memory at address 0x1>)
    at generated/gui/ice40/gui_ice40_autogen/7FXS3DRWOX/moc_worker.cpp:790
#9  0x000000000d9e2291 in QMetaCallEvent::placeMetaCall(QObject*) ()
   from C:\msys64\mingw64\bin\Qt5Core.dll
#10 0x000000000d9e63ea in QObject::event(QEvent*) ()
   from C:\msys64\mingw64\bin\Qt5Core.dll
#11 0x0000000066207c2c in QApplicationPrivate::notify_helper(QObject*, QEvent*) () from C:\msys64\mingw64\bin\Qt5Widgets.dll
#12 0x000000006620ee83 in QApplication::notify(QObject*, QEvent*) ()
   from C:\msys64\mingw64\bin\Qt5Widgets.dll
#13 0x0000000000458583 in nextpnr_ice40::Application::notify (
    this=<optimized out>, receiver=<optimized out>, event=<optimized out>)
    at ../gui/application.cc:41
#14 0x000000000d9ba1c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*)
    () from C:\msys64\mingw64\bin\Qt5Core.dll
#15 0x000000000d9c0529 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () from C:\msys64\mingw64\bin\Qt5Core.dll
#16 0x000000006dd30a9e in qwindows!qt_plugin_instance ()
   from C:\msys64\mingw64\share\qt5\plugins\platforms\qwindows.dll
#17 0x000000000da12df9 in QEventDispatcherWin32Private::sendTimerEvent(int) ()
   from C:\msys64\mingw64\bin\Qt5Core.dll
#18 0x0000000076e49bbd in USER32!TranslateMessageEx ()
   from C:\Windows\system32\user32.dll
#19 0x0000000076e498c2 in USER32!TranslateMessage ()
   from C:\Windows\system32\user32.dll
#20 0x000000000da121cb in QEventDispatcherWin32::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () from C:\msys64\mingw64\bin\Qt5Core.dll
#21 0x000000006dd30a85 in qwindows!qt_plugin_instance ()
   from C:\msys64\mingw64\share\qt5\plugins\platforms\qwindows.dll
#22 0x000000000d9b8a7f in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () from C:\msys64\mingw64\bin\Qt5Core.dll
#23 0x000000000d9c19a7 in QCoreApplication::exec() ()
   from C:\msys64\mingw64\bin\Qt5Core.dll
#24 0x00000000005bf365 in main (argc=<optimized out>, argv=0xd5c4f60)
    at ../ice40/main.cc:406
```

q3k, 2018-08-01:
@cr1901 Please try https://github.com/YosysHQ/nextpnr/commit/101de5b7151e996b026fa3705fc5ec5d065142f3  - we had quite a bug that might've caused this - but we're not yet sure.

cr1901, 2018-08-01:
101de5b did _not_ fix the packing issue :(...

q3k, 2018-08-01:
Handing this off to @mmicko , as I can't seem to be able to replicate this on Linux.

mmicko, 2018-08-01:
ok will build with mingw on windows now to test

mmicko, 2018-08-01:
Was not able to reproduce it, but will try updating msys2 packages to latest and try again.

mmicko, 2018-08-01:
@cr1901 had a msys2 with mingw gcc 7.x installed and on it I could not reproduce issue. Have tried updating all packages to latest and doing clean compile and still could not reproduce your issue. If you can try updating your msys2 environment, it would not be first time that compile was broken in some of updates.

cr1901, 2018-08-01:
@mmicko This did not fix the problem- backtrace: http://ix.io/1iXy

mmicko, 2018-08-02:
Thanks for the input.  Think I am now closer to solution for your issue. Managed to reproduce it in MSVC as well in debug mode only.  But it makes sense.

cr1901, 2018-08-02:
@mmicko Bug is now fixed on my end... I can now run P&R from the GUI successfully :).
[image]

closed 2018-08-02 by cr1901