YosysHQ/nextpnr #4: Loading Sample Blinky Project From GUI Causes SIGTRAP on MinGW64 Windows

cr1901, 2018-08-01:
## Environment
```
$ g++ -v
Using built-in specs.
COLLECT_GCC=C:\msys64\mingw64\bin\g++.exe
COLLECT_LTO_WRAPPER=C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/7.3.0/lto-wrapper.exe
Target: x86_64-w64-mingw32
Configured with: ../gcc-7.3.0/configure --prefix=/mingw64 --with-local-prefix=/mingw64/local --build=x86_64-w64-mingw32 --host=x86_64-w64-mingw32 --target=x86_64-w64-mingw32 --with-native-system-header-dir=/mingw64/x86_64-w64-mingw32/include --libexecdir=/mingw64/lib --enable-bootstrap --with-arch=x86-64 --with-tune=generic --enable-languages=c,lto,c++,objc,obj-c++,fortran,ada --enable-shared --enable-static --enable-libatomic --enable-threads=posix --enable-graphite --enable-fully-dynamic-string --enable-libstdcxx-time=yes --enable-libstdcxx-filesystem-ts=yes --disable-libstdcxx-pch --disable-libstdcxx-debug --disable-isl-version-check --enable-lto --enable-libgomp --disable-multilib --enable-checking=release --disable-rpath --disable-win32-registry --disable-nls --disable-werror --disable-symvers --with-libiconv --with-system-zlib --with-gmp=/mingw64 --with-mpfr=/mingw64 --with-mpc=/mingw64 --with-isl=/mingw64 --with-pkgversion='Rev2, Built by MSYS2 project' --with-bugurl=https://sourceforge.net/projects/msys2 --with-gnu-as --with-gnu-ld
Thread model: posix
gcc version 7.3.0 (Rev2, Built by MSYS2 project)

$ nextpnr-ice40 -V
nextpnr-ice40 -- Next Generation Place and Route (git sha1 2743d0f)
```

## Steps to Duplicate
1. Run `nextpnr-ice40 --gui` from MSYS2/MinGW64 `bash`
2. File > Open > blinky.proj

(For the file dialog, navigate to the `ice40` directory inside the `nextpnr` source tree)

## Expected Behavior
No SIGTRAP.

## Actual Behavior
Application immediately closes with "Segmentation Fault".

Backtrace/SIGTRAP source is here: http://ix.io/1iV0

q3k, 2018-08-01:
I can replicate this on Linux, it doesn't seem like a MinGW issue.

q3k, 2018-08-01:
What seems to be happening is that for some reason when loading a project TreeModel::Model::type() gets called on an index that doesn't hold a valid internal pointer to a TreeModel::Item.

```nodeFromIndex-> internal (0x55954d71b9c0)
flags(0x55954d71b9c0)
Segmentation fault```

q3k, 2018-08-01:
Hum, seems like this issue is actually unrelated to the crash that I'm observing.  Hold on.

q3k, 2018-08-01:
@cr1901 Please try https://github.com/YosysHQ/nextpnr/commit/101de5b7151e996b026fa3705fc5ec5d065142f3  - we had quite a bug that might've caused this - but we're not yet sure.

cr1901, 2018-08-01:
Blinky project load was fixed by 101de5b!

closed 2018-08-01 by q3k