YosysHQ/nextpnr #7: ice40: "Floating wire node value" message is mangled

whitequark, 2018-08-01:
Looks something like this:
```
Info: Importing module top
Info:       Floating wire node value, 'x' of port 'MASK' in cell 'storage.0.0.0' of module 'top'
, converted to zero driverInfo:       Floating wire node value, 'x' of port 'MASK' in cell 'storage.0.0.0' of module 'top'
, converted to zero driverInfo:       Floating wire node value, 'x' of port 'MASK' in cell 'storage.0.0.0' of module 'top'
, converted to zero driverInfo:       Floating wire node value, 'x' of port 'MASK' in cell 'storage.0.0.0' of module 'top'
, converted to zero driverInfo:       Floating wire node value, 'x' of port 'MASK' in cell 'storage.0.0.0' of module 'top'
, converted to zero driverInfo:       Floating wire node value, 'x' of port 'MASK' in cell 'storage.0.0.0' of module 'top'
, converted to zero driverInfo:       Floating wire node value, 'x' of port 'MASK' in cell 'storage.0.0.0' of module 'top'
... etc
```
[Design](https://github.com/YosysHQ/nextpnr/files/2250364/glasgow_ci9sc9wy.zip)

daveshah1, 2018-08-01:
Thanks! Should be fixed in 1d6ec0f

q3k, 2018-08-04:
@whitequark Would you mind verifying that this got fixed?

whitequark, 2018-08-04:
I haven't seen it again, so I believe it's fixed.

closed 2018-08-04 by whitequark