YosysHQ/nextpnr #9: Supported input formats

olofk, 2018-08-01:
More of a couple of questions than an issue.

Is the json format the only intended input format? I see that new versions of yosys can generate json, but up until now I have used blif.

Will there be any support for edf?

Is there some command I'm missing to convert blif/edf to json?

dvc94ch, 2018-08-01:
Mmh I thought json was supported for a long time. New versions of yosys can read json which is cool. Can't you do something like `yosys -p read_blif;write_json`?

cliffordwolf, 2018-08-02:
> Can't you do something like `yosys -p read_blif; write_json`?

This.

Yosys does already support converting a bunch of different netlist formats, no need to replicate this functionality in nextpnr. The main advantage of JSON of BLIF is that JSON preserves much more of the information in a design. Going `-> BLIF -> JSON` works, but you will lose information such as cell names because cell names don't exist in the standard BLIF file format.

closed 2018-08-02 by cliffordwolf