YosysHQ/nextpnr #17: GUI startup segmentation fault on OS X

derekmulcahy, 2018-08-02:
On OS X 10.13.6 (High Sierra) I get a segmentation fault when running nextpnr-ice40 (0208897aa3b5bbe91810e728c825566f87c6fa0c).

I am running the example from the GitHub repo.

`nextpnr-ice40 --json blinky.json --pcf blinky.pcf --asc blinky.asc --gui`

I can "fix" the issue by commenting out line basewindow.cc:81

`centralTabWidget->tabBar()->tabButton(0, QTabBar::RightSide)->resize(0, 0);`

Poking about with LLDB indicates that tabButton() is returning NULL for QTabBar::RightSide. If it is passed LeftSide then a value is returned.

Here is the stack trace for the issue.
```
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x8)
  * frame #0: 0x0000000109c6bc30 QtWidgets`QWidget::resize(QSize const&) + 16
    frame #1: 0x00000001000c36de nextpnr-ice40`nextpnr_ice40::BaseMainWindow::BaseMainWindow(std::__1::unique_ptr<nextpnr_ice40::Context, std::__1::default_delete<nextpnr_ice40::Context> >, QWidget*) [inlined] QWidget::resize(this=<unavailable>, w=0, h=0) at qwidget.h:838 [opt]
    frame #2: 0x00000001000c36ca nextpnr-ice40`nextpnr_ice40::BaseMainWindow::BaseMainWindow(this=0x00007ffeefbff150, context=<unavailable>, parent=<unavailable>) at basewindow.cc:81 [opt]
    frame #3: 0x00000001000e91b2 nextpnr-ice40`nextpnr_ice40::MainWindow::MainWindow(this=0x00007ffeefbff150, context=<unavailable>, args=(type = HX1K, package = "tq144"), parent=<unavailable>) at mainwindow.cc:40 [opt]
    frame #4: 0x0000000100097e07 nextpnr-ice40`main(argc=<unavailable>, argv=<unavailable>) at main.cc:396 [opt]
    frame #5: 0x00007fff6762e015 libdyld.dylib`start + 1
```

derekmulcahy, 2018-08-02:
From  [StackOverflow](https://stackoverflow.com/questions/2616483/close-button-only-for-some-tabs-in-qt) this is a difference between OS X and other systems. Some lean to the RightSide, Macs lean to the LeftSide. Looks like I am your Mac GUI tester :-)

mmicko, 2018-08-02:
Thanks for report. Fixed with https://github.com/YosysHQ/nextpnr/commit/6c99f7525e979d58286a18c819a692d16c270217
Have tried on High Sierra as well, but please confirm

derekmulcahy, 2018-08-02:
That fixed it, thanks.

closed 2018-08-02 by derekmulcahy