YosysHQ/nextpnr #24: GUI CPU usage

cliffordwolf, 2018-08-03:
Just running `../nextpnr-ice40 --hx8k --gui` (without loading a design or doing anything) already has a pretty high CPU usage.

q3k, 2018-08-03:
@cliffordwolf out-of-band you told me it's especially bad on Nvidia/Linux. Would you mind providing an apitrace so I can look at what's the performance bottleneck in that case?

cliffordwolf, 2018-08-04:
apitrace: http://scratch.clifford.at/nextpnr-ice40.trace.xz

datenwolf, 2018-09-19:
@cliffordwolf since I told you OOB that I'd like to pull my weight on graphics/OpenGL issues, I've started hacking on this stuff.

There's a lot that can be improved in the OpenGL code performance wise: Lot of redundant copies, that can be avoided, picking could be performed through a ID buffer populated by the GPU while drawing (picking reduces to indexing into the generated buffer, reading the ID of the element therein). Selecting the drawing style (hover, grayed out, etc.) can be done entirely in the shader, changing the semantics of the graphics element list to allow for direct access by the GPU instead of requiring a coalesce copy each frame redraw…

q3k, 2018-09-19:
Absolutely, the current renderer code can be improved quite a lot - it grew organically, and the first iteration of the shader code had quite a different scope in mind than now.

I don't currently have time to work on this, so go ahead :).

datenwolf, 2018-09-19:
Already on it. Just want to make sure, we don't step on each others' toes.

smunaut, 2020-03-04:
Any update on this ?  Starting an ECP5 GUI makes makes my fan spin ... it not that much CPU (not that I can see in `top` at least) I think it's a lot of wake ups / small cycles ?  Even when it's running on another virtual desktop it slows down my system.

mmicko, 2020-03-04:
@smunaut There was no updates to GUI code coming from outside contributors so I will work on solving this issue. There is a lot of space for improvement, and also ECP5 is not that very large so performance issues should better be solved until we start supporting some larger FPGA.

datenwolf, 2020-03-04:
I had to reprioritize my health (i.e. spend more time away from screens and with human beings) over adding that to my workload. Maybe I'll look at this again, sometime again.

mmicko, 2020-03-04:
@datenwolf no worries there will work on this issue. There will always be place for new ideas and optimizations :)