YosysHQ/nextpnr #49: Negative integer in JSON causes error

eddiehung, 2018-08-14:
This snippet of the attached JSON (on line 21643, generated by Yosys):
```
        "$abc$529$auto$blifparse.cc:465:parse_blif$538": {
          "hide_name": 1,
          "type": "LUT5",
          "parameters": {
            "INIT": -2147483648
          },
```
causes nextpnr to issue the following error:
```
ERROR: Unexpected character in JSON file: '-'
ERROR: Loading design failed.
```

[blinky.json.txt](https://github.com/YosysHQ/nextpnr/files/2287680/blinky.json.txt)

Tasks:
- Fix this
- Provide better error messages, e.g. a line number

I dug into Yosys, and it's policy is that if the number of bits in a parameter (`INIT`) is less than 32 bits, then its value is output as an integer. Any more, then a string is used. Whether `INIT` can/should be marked as an unsigned integer in Yosys may be something that @cliffordwolf can chime in...

q3k, 2018-08-14:
This is generally why Protobufs would make for a better interchange format than JSON: not having to deal with batshit insane underspecified number formatting rules [1]...

[1] - http://seriot.ch/parsing_json.php#22

ZipCPU, 2018-08-14:
Please check out the json-update branch and see if that fixes your error.
You can see the changes [here](https://github.com/YosysHQ/nextpnr/compare/json-update).

Dan

eddiehung, 2018-08-15:
@ZipCPU Thanks Dan! Your branch now accepts the previously attached file. 

I changed the single minus sign into two minus signs to try and create an error to trigger a new error, and now it gives:
```
ERROR: Unexpected character in JSON file, line 21645: '}'
```
which is the line following the actual error. Is this what you were expecting?

In other news, @cliffordwolf fixed upstream Yosys with this commit: https://github.com/YosysHQ/yosys/commit/1dd156f5167d9949eccb7abcbc65d9bc74f49ee8 so this shouldn't be an issue moving forward.

ZipCPU, 2018-08-15:
Yes, somewhat.  Two minus signs creates a zero followed by the negative number in question.

eddiehung, 2018-08-18:
Resolved by #51

closed 2018-08-18 by eddiehung