YosysHQ/nextpnr #50: Speedup placer by performing incremental cost updates

eddiehung, 2018-08-14:
Based on [this profile](https://github.com/YosysHQ/nextpnr/files/2276516/design03.pdf) it appears that 86% of total runtime is spent inside placement, and 47% of total runtime is spent inside `get_net_metric()` which is responsible for computing the cost of a swap.

Instead of computing the half-perimeter-wire-length (HPWL) and worst-slack from scratch for every pin of every net of every affected cell, perhaps an incremental update can be performed? For example:
- if a pin is moved entirely inside the existing HPWL, then there's no need to recompute scratch
- if a pin does not currently have the worst-slack of the net, and does not increase the worst-slack, then no need to recompute from scratch
- etc.

daveshah1, 2019-05-18:
#219 adds initial incremental bounding box computation (as placer1 performance is important to fast post-HeAP refinement) and #263 improves it further by considering more cases where a full update isn't needed. The placer now uses per-arc criticality/slack rather than worst slack in its timing model, so incremental slack updates would be more complicated.

gatecat, 2022-08-18:
closing as the above mentioned fixes now cover most of this

closed 2022-08-18 by gatecat