YosysHQ/nextpnr #68: Slower timing results with higher frequency target

esden, 2018-09-04:
I am building picosoc for ice40up5k and getting strange results.

When running nextpnr with a 13mhz target it produces an asc with 10.91 MHz fmax according to nextpnr and 11.99 MHz according to icetime. On the other hand when running nextpnr with 12mhz target it produces an asc with 12.05 MHz fmax according to nextpnr and 12.15 MHz according to icetime.
I have uploaded a bunch of files including output logs, and asc files for reference: http://esden.net/blubb/picosoc/2018_09_04/
Yosys version: `Yosys 0.7+650 (git sha1 4d269f9b, clang 6.0.0-1ubuntu2 -fPIC -Os)`
nextpnr version: `nextpnr-ice40 -- Next Generation Place and Route (git sha1 61b1a75)`
icestorm version built from: sha 52f91d92

Let me know if you need any additional information.

q3k, 2018-09-04:
@eddiehung wanna take a look at this?

eddiehung, 2018-09-04:
There are a few things at play here:
- Simulated Annealing (SA) is an inherently deterministically-random process, such that making small changes in the design or the constraints can lead to wildly different solutions
- The algorithms used in nextpnr are currently pretty basic, and there is definitely more room for improvement (e.g. see #27)
- There is a difference between icetime results and nextpnr because they use different timing models

esden, 2018-09-05:
@daveshah1 yes you are correct. By running nextpnr with different seed values it does result in different fmax. I ran a few seeds:

```
Seed	fmax
	1	10.57MHz
	2	11.38MHz
	3	11.60MHz
	4	12.56MHz
	5	12.51MHz
	6	11.76MHz
	7	10.79MHz
	8	11.61MHz
	9	11.67MHz
	10	11.67MHz
	11	11.08MHz
	12	11.48MHz
```

I am aware that this issue might not be possible to resolve. In that case I think there should be a mechanism to allow for a "search" run. This means running a defined amount of times using random seeds and find one with the best result. Maybe this could even be parallelized to allow for faster search? Of course I can just use a shell or python script to build such a thing but if this is an issue that people will run into on a regular basis it might be more practical to be provided through the command line parameters of nextpnr.

@eddiehung thank you for looking into this. Let me know if you want me to test something or need additional data.

closed 2023-09-13 by gatecat