YosysHQ/nextpnr #69: Using `PLLOUTGLOBAL` on PLL results in :Assertion failure"

cr1901, 2018-09-04:
## Actual Behavior

The following Verilog module:

```
William@William-THINK MINGW64 ~/src/tinyfpga-soc
$ cat pll-crash/build/top.v
/* Machine-generated using Migen */
module top(
        input clk16,
        output out
);

wire sys_clk;
wire clk16_1;
wire clk32;

// Adding a dummy event (using a dummy signal 'dummy_s') to get the simulator
// to run the combinatorial process once at the beginning.
// synthesis translate_off
reg dummy_s;
initial dummy_s <= 1'd0;
// synthesis translate_on

assign clk16_1 = clk16;
assign out = sys_clk;

SB_PLL40_CORE #(
        .DIVF(6'd63),
        .DIVQ(3'd5),
        .DIVR(1'd0),
        .FILTER_RANGE(3'd1)
) SB_PLL40_CORE (
        .REFERENCECLK(clk16_1),
        .RESETB(1'd1),
        .PLLOUTGLOBAL(clk32)
);

SB_GB SB_GB(
        .USER_SIGNAL_TO_GLOBAL_BUFFER(clk32),
        .GLOBAL_BUFFER_OUTPUT(sys_clk)
);

endmodule
```

results in the following assertion failure:

```
William@William-THINK MINGW64 ~/src/tinyfpga-soc/pll-crash
$ python3 pll-crash.py
Warning: Found one of those horrible `(synopsys|synthesis) translate_off' comments.
Yosys does support them but it is recommended to use `ifdef constructs instead!
Info: Importing module top
Info: Rule checker, Verifying pre-placed design
Info: Checksum: 0x191aa3eb

Info: constrained 'clk16' to bel 'X9/Y33/io0'
Info: constrained 'out' to bel 'X0/Y30/io0'

Info: Packing constants..
Info: Promoting globals..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info: Packing non-LUT FFs..
Info: Packing carries..
Info: Packing RAMs..
Info: Packing special functions..
Warning: PLL 'SB_PLL40_CORE' has unknown connected port 'PLLOUTGLOBAL' - ignoring
Info:   constrained PLL 'SB_PLL40_CORE_PLL' to X16/Y0/pll_3
Info:   constrained PLL 'SB_PLL40_CORE_PLL' to X16/Y33/pll_3
Info: Constraining chains...
Info: Checksum: 0x28c010d3

Info: Annotating ports with timing budgets for target frequency 16.00 MHz

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
terminate called after throwing an instance of 'nextpnr_ice40::assertion_failure'
  what():  Assertion failure: port_fanin.empty() (../common/timing.cc:167)
```

## Expected behavior
`PLLOUTGLOBAL` followed by `SB_GB` should be supported in `nextpnr`, even if it's an obsolete hint (?) to the `pnr` tool.

## Other information
```
William@William-THINK MINGW64 ~/src/tinyfpga-soc
$ uname -a
MINGW64_NT-6.1 William-THINK 2.10.0(0.325/5/3) 2018-02-09 15:25 x86_64 Msys
```

## MCVE
I have provided an MCVE: [pll-crash.zip](https://github.com/YosysHQ/nextpnr/files/2350381/pll-crash.zip)

q3k, 2018-09-04:
First of all, whoops, [these branches should be swapped](https://github.com/YosysHQ/nextpnr/blob/master/ice40/pack.cc#L754-L759) so that we fail early instead of crashing later.

Second of all, seems like PLLOUTGLOBAL is not a real Bel port, same as PLLOUTCORE. I think we should implement designs that use this port using the following logic:

- If the out port PLLOUTGLOBAL is used, **THEN** remap the port to PLLOUT_A
  - **AND** if the PLLOUT_A port drives any cell that is not a SB_GB
  - **THEN** insert an SB_GB between the PLLOUT_A and downhill cell ports

Similar to what we do with LOCK and ICESTORM_LCs...

@daveshah1 , what do you think?

daveshah1, 2018-09-04:
PLLOUTGLOBAL has a dedicated connection to a global network via the padin, as if an SB_GB_IO was placed at the relevant location. This isn't the same as following the PLL output with a SB_GB (in arachne and the vendor tools, PLLOUTGLOBAL followed by SB_GB will waste a precious global buffer, and this is correct behaviour imo).

The proper solution is to add support for padin, which requires small changes to the bitstream gen as these are "extra bits" not part of a tile, implement PLLOUTGLOBAL properly and SB_GB_IO at the same time.

mmicko, 2018-09-21:
Have added a temporary fix for this, since some additional things needed to be added for generated bitstream. Can you please check if https://github.com/YosysHQ/nextpnr/pull/76 works for you ?

mmicko, 2018-09-22:
> @mmicko <https://github.com/mmicko> This solves the crash, but then I get
> the following warning:
>
> Warning: PLL 'SB_PLL40_CORE' is using port PLLOUTGLOBAL but implementation does not actually use the global clock output of the PLL
>
> Does *the implementation* refer to nextpnr or my design? Because I route
> PLLOUTGLOBAL directly to my intended system clock.

Warning is due to missing part in implementation, afraid it needs to wait
for David to be fully resolved. As I understood it should even work with
this change but not fully.

mmicko, 2018-09-23:
Agree. It seams your other PLL issue is connected to global buffer bitstream handling as well.

smunaut, 2018-11-16:
I've updated my tree with my latest changes, cleaning it all up. I think it's ready to be reviewed / tested.

For the IO config what I ended up doing is :
 - Do a prescan of all the cells to find the PLL and collects the SB_IO that are used by that PLL.
   Either as clock input because it's a PAD type PLL. Or for the output. Note that it doesn't matter if there is anything actually connected to the output port or not, it's purely the PLL type that determines if an SB_IO is unavailable because of the PLL. (i.e. 1 or 2 outputs). Even if the output is global, the SB_IO is still unavailable.
 - During the normal scan of the cells, for SB_IO, if a cell is used as PAD input, we make sure to enable its input buffer. If the cell is used as part of the output path of the pll we make sure the user isn't trying to use the input as well. In both cases, we _do_not_ configure PINTYPE[1:0] there, we omit them.
 - During the normal scan of the cell, for the PLL, we take all the SB_IO locations used for the output path and we make sure to set the PINTYPE[1:0]  appropriately.

So the IE/REN bits are always configured as part of the SB_IO config. That's because it's only needed if that site is used as an actual input (either pad or otherwise) and in that case, there will be a SB_IO cell in the list of cells for that site.

The PINTYPE bits are configured during the PLL config because for some of the SB_IO used in the output path, it's possible we don't have any SB_IO that correspond to that site in the design.

PR #131 (2018-11-20): iCE40: Bug fixes and general improvement of global network support
* The first part of the PR are various bug fixes and improvements.
* Then follows a bunch of commits to improve 
  * First all the requires infrastructure and chipdb changes
  * Then the helper that will we used to create the 'forPadIn' SB_GB we use for this and to handle the extra_bits in the bitstream
  * Finally for each cell that support a global output, add support for it with the helper
* The final commits adds support for the RGB led driver found in the UP5k devices

Submitted as 1 PR because of the dependencies on some  [... cut]
closed 2018-11-20 by daveshah1