YosysHQ/nextpnr #70: SB_RGBA_DRV support

smunaut, 2018-09-05:
Trying a design using the RGB drivers result in a fatal error : 

> ERROR: no timing info for port 'RGB0' of cell type 'SB_RGBA_DRV'

In arachne, there is a warning but it proceeds. Here this is preventing the process altogether.

widgetlords, 2018-09-05:
I believe this is related: I get the same error but with the WKUP pin of the SPI hard modules and also for the I2C hard modules
```ERROR: no timing info for port 'SDAOE' of cell type 'SB_I2C'```

eddiehung, 2018-09-06:
Thanks for the report chaps! This is an intentional hard failure to catch uses of timing arcs that haven't been populated yet -- perhaps @daveshah1 can help fill those in?

eddiehung, 2018-09-07:
Okay, so the issue appears to be more than just timing arcs. Standby!

darkfader, 2018-09-07:
"No wire found for port RGB0" ;)

smunaut, 2018-10-24:
@eddiehung Standing by ... for a while now :P

Is there an easy way to just ignore the lack of timing ?

daveshah1, 2018-10-24:
The problem is that the primitive isn't supported at all, lack of timing is just the first error that occurs. I'll see if I have time to get this done in the next few weeks.

smunaut, 2018-11-20:
Fixed

closed 2018-11-20 by smunaut