YosysHQ/nextpnr #72: ice40: PACKER_VCC appears on critical path

whitequark, 2018-09-15:
E.g. this report, trivial to reproduce:
```
Info: Critical path report:
Info: curr total
Info:  0.0  0.0  Source $PACKER_VCC.O
Info:  6.3  6.3    Net $PACKER_VCC_NET budget 2.777000 ns (11,1) -> (12,31)
Info:                 Sink $gbuf_$PACKER_VCC_NET_$glb_clk.USER_SIGNAL_TO_GLOBAL_BUFFER
Info:  0.0  6.3  Source $gbuf_$PACKER_VCC_NET_$glb_clk.GLOBAL_BUFFER_OUTPUT
Info:  6.8 13.1    Net $PACKER_VCC_NET_$glb_clk budget 2.777000 ns (12,31) -> (10,28)
Info:                 Sink $auto$alumacc.cc:474:replace_alu$1660.slice[1].carry$CARRY.I1
(etc)
```
I understand that it has the enormous 6.3 ns delay because it's a single `1'b1` net for the entire chip, but that shouldn't matter if you have a reset, and so I think it shouldn't contribute to timing.

eddiehung, 2018-09-15:
Interesting, and you are absolutely right! I had thought about this possibility but never saw it during testing so it dropped by the way side.

It should be an easy fix: I can add a check to see if there are no inputs to the BLE, then timing analysis should ignore it.

whitequark, 2018-09-15:
Here is a full repro then: [spi-flash-avr.zip](https://github.com/YosysHQ/nextpnr/files/2386088/spi-flash-avr.zip).

eddiehung, 2018-09-15:
Confirm repro on my end. Working on it!

eddiehung, 2018-09-28:
@whitequark does my branch resolve your issue?

eddiehung, 2018-10-10:
Wunderbar. PR#88 submitted.

eddiehung, 2018-10-11:
#88 merged.

closed 2018-10-11 by eddiehung