YosysHQ/nextpnr #73: Critical paths with multiple global buffers

cliffordwolf, 2018-09-16:
I see critical paths like the following:

```
Info: Critical path report:
Info: curr total
Info:  1.4  1.4  Source $auto$simplemap.cc:420:simplemap_dff$2753_DFFLC.O
Info:  1.8  3.2    Net qmem_done budget 0.618000 ns (12,9) -> (12,9)
Info:                 Sink $abc$14511$auto$blifparse.cc:492:parse_blif$14726_LC.I3
Info:  0.0  3.2  Source $abc$14511$auto$blifparse.cc:492:parse_blif$14726_LC.O
Info:  5.0  8.1    Net mem_client_qmem budget 1.856000 ns (12,9) -> (13,31)
Info:                 Sink $gbuf_mem_client_qmem_$glb_clk.USER_SIGNAL_TO_GLOBAL_BUFFER
Info:  0.0  8.1  Source $gbuf_mem_client_qmem_$glb_clk.GLOBAL_BUFFER_OUTPUT
Info:  1.8  9.9    Net mem_client_qmem_$glb_clk budget 1.856000 ns (13,31) -> (13,6)
Info:                 Sink $abc$14511$auto$blifparse.cc:492:parse_blif$14728_LC.I3
Info:  0.9 10.8  Source $abc$14511$auto$blifparse.cc:492:parse_blif$14728_LC.O
Info:  2.8 13.6    Net mem_addr[0] budget 1.856000 ns (13,6) -> (10,6)
Info:                 Sink $abc$14511$auto$blifparse.cc:492:parse_blif$15156_LC.I0
Info:  1.3 14.9  Source $abc$14511$auto$blifparse.cc:492:parse_blif$15156_LC.O
Info:  2.3 17.2    Net $abc$14511$n1856 budget 2.386000 ns (10,6) -> (8,6)
Info:                 Sink $auto$alumacc.cc:474:replace_alu$1146.slice[0].adder_LC.I2
Info:  0.0 17.2  Source $auto$alumacc.cc:474:replace_alu$1146.slice[0].adder_LC.O
Info:  6.8 24.0    Net mem.shamt[0] budget 2.386000 ns (8,6) -> (19,31)
Info:                 Sink $gbuf_mem.shamt[0]_$glb_clk.USER_SIGNAL_TO_GLOBAL_BUFFER
Info:  0.0 24.0  Source $gbuf_mem.shamt[0]_$glb_clk.GLOBAL_BUFFER_OUTPUT
Info:  4.4 28.4    Net mem.shamt[0]_$glb_clk budget 2.394000 ns (19,31) -> (22,1)
Info:                 Sink $abc$14511$auto$blifparse.cc:492:parse_blif$14739_LC.I0
Info:  1.3 29.7  Source $abc$14511$auto$blifparse.cc:492:parse_blif$14739_LC.O
Info:  2.3 32.0    Net mem.shifted_wen[1] budget 2.394000 ns (22,1) -> (24,1)
Info:                 Sink $abc$14511$auto$blifparse.cc:492:parse_blif$14741_LC.I0
Info:  0.0 32.0  Source $abc$14511$auto$blifparse.cc:492:parse_blif$14741_LC.O
Info:  3.0 34.9    Net $abc$14511$n479 budget 2.394000 ns (24,1) -> (25,0)
Info:                 Sink mem.ram[0].spram_RAM.WREN
```

Maybe we should make sure not to place global buffers in the input- or output-cone of another global when promoting signals. (Also, we should probably provide an option to disable automatic gbuf promotion of logic signal.)

eddiehung, 2018-09-16:
Interesting stuff! An option to control global buffer promotion is definitely an idea:
- Promote clocks and other signals
- Promote clocks only
- No promotion

However, what do you think the right thing is? Should every path should only contain at most one global buffer? If so, what should the priority be:
- First one to be promoted
- Highest fanout
- or something else?

Alternatively, should we consider (maybe at routing) whether the best thing is to use the faster of the global output, or to use regular routing?

mithro, 2018-09-16:
Why not just use a normal cost model?

cliffordwolf, 2018-09-17:
> Alternatively, should we consider (maybe at routing) whether the best thing is to use the faster of the global output, or to use regular routing?

I also thought about that already. My main concern with this approach is that placing wouldn't be aware of the global net, thus it would assign value to moving source and sinks together, but it would assign no value to moving the source close to the global buffer driver.

I think the best approach would be to have some switches to turn promotion and and off explicitly for different classes of nets, and add some improvements to the automatic promotion of logic signals. As a simple heuristic it probably is never good to have two gbufs in the longest path. Better would probably be to use the timing budgets to inform the decision where inserting a gbuf does hurt timing more than it benefits routability.

closed 2019-02-05 by daveshah1