YosysHQ/nextpnr #77: Disable readline support?

mithro, 2018-09-19:
Is there a way to disable building with readline support? Is `-DENABLE_READLINE=No` enough?

mmicko, 2018-09-21:
There is no direct dependency to readline in nextpnr. Can you give a hint to what do you refer ?

mithro, 2018-09-21:
Hrm, conda was detecting that nextpnr was being linked against libreadline but I may be mistaken.

daveshah1, 2018-09-21:
This might be coming through the dependency on libpython? In that case, I don't think you could disable it without rebuilding libpython.

daveshah1, 2018-11-26:
`ldd -v nextpnr-ice40` prints for me:

```
	linux-vdso.so.1 (0x00007fffafda7000)
	libboost_python37.so.1.68.0 => /usr/lib/libboost_python37.so.1.68.0 (0x00007f1184377000)
	libboost_filesystem.so.1.68.0 => /usr/lib/libboost_filesystem.so.1.68.0 (0x00007f1184359000)
	libboost_thread.so.1.68.0 => /usr/lib/libboost_thread.so.1.68.0 (0x00007f118432c000)
	libboost_program_options.so.1.68.0 => /usr/lib/libboost_program_options.so.1.68.0 (0x00007f11842a4000)
	libboost_system.so.1.68.0 => /usr/lib/libboost_system.so.1.68.0 (0x00007f118429e000)
	libboost_chrono.so.1.68.0 => /usr/lib/libboost_chrono.so.1.68.0 (0x00007f1184293000)
	libboost_date_time.so.1.68.0 => /usr/lib/libboost_date_time.so.1.68.0 (0x00007f118427d000)
	libboost_atomic.so.1.68.0 => /usr/lib/libboost_atomic.so.1.68.0 (0x00007f1184278000)
	libpthread.so.0 => /usr/lib/libpthread.so.0 (0x00007f1184257000)
	libQt5OpenGL.so.5 => /usr/lib/libQt5OpenGL.so.5 (0x00007f11841fe000)
	libGL.so.1 => /usr/lib/libGL.so.1 (0x00007f118416b000)
	libGLU.so.1 => /usr/lib/libGLU.so.1 (0x00007f1183eeb000)
	libpython3.7m.so.1.0 => /usr/lib/libpython3.7m.so.1.0 (0x00007f1183b70000)
	libQt5Widgets.so.5 => /usr/lib/libQt5Widgets.so.5 (0x00007f118350e000)
	libQt5Gui.so.5 => /usr/lib/libQt5Gui.so.5 (0x00007f1182f7d000)
	libQt5Core.so.5 => /usr/lib/libQt5Core.so.5 (0x00007f1182a7b000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x00007f11828ec000)
	libm.so.6 => /usr/lib/libm.so.6 (0x00007f1182767000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x00007f118274b000)
	libc.so.6 => /usr/lib/libc.so.6 (0x00007f1182587000)
	librt.so.1 => /usr/lib/librt.so.1 (0x00007f118257d000)
	libdl.so.2 => /usr/lib/libdl.so.2 (0x00007f1182578000)
	libutil.so.1 => /usr/lib/libutil.so.1 (0x00007f1182573000)
	/lib64/ld-linux-x86-64.so.2 => /usr/lib64/ld-linux-x86-64.so.2 (0x00007f118fa19000)
	libGLX.so.0 => /usr/lib/libGLX.so.0 (0x00007f118253e000)
	libX11.so.6 => /usr/lib/libX11.so.6 (0x00007f11823ff000)
	libXext.so.6 => /usr/lib/libXext.so.6 (0x00007f11821ed000)
	libGLdispatch.so.0 => /usr/lib/libGLdispatch.so.0 (0x00007f1182131000)
	libpng16.so.16 => /usr/lib/libpng16.so.16 (0x00007f11820f9000)
	libharfbuzz.so.0 => /usr/lib/libharfbuzz.so.0 (0x00007f1182016000)
	libz.so.1 => /usr/lib/libz.so.1 (0x00007f1181dfd000)
	libsystemd.so.0 => /usr/lib/libsystemd.so.0 (0x00007f1181d71000)
	libicui18n.so.63 => /usr/lib/libicui18n.so.63 (0x00007f1181a93000)
	libicuuc.so.63 => /usr/lib/libicuuc.so.63 (0x00007f11818c3000)
	libpcre2-16.so.0 => /usr/lib/libpcre2-16.so.0 (0x00007f1181849000)
	libdouble-conversion.so.3.0.0 => /usr/lib/libdouble-conversion.so.3.0.0 (0x00007f1181832000)
	libglib-2.0.so.0 => /usr/lib/libglib-2.0.so.0 (0x00007f118170d000)
	libxcb.so.1 => /usr/lib/libxcb.so.1 (0x00007f11816e3000)
	libfreetype.so.6 => /usr/lib/libfreetype.so.6 (0x00007f118141a000)
	libgraphite2.so.3 => /usr/lib/libgraphite2.so.3 (0x00007f11813f5000)
	liblzma.so.5 => /usr/lib/liblzma.so.5 (0x00007f11811cf000)
	liblz4.so.1 => /usr/lib/liblz4.so.1 (0x00007f11811ae000)
	libgcrypt.so.20 => /usr/lib/libgcrypt.so.20 (0x00007f118108f000)
	libicudata.so.63 => /usr/lib/libicudata.so.63 (0x00007f117f6a1000)
	libpcre.so.1 => /usr/lib/libpcre.so.1 (0x00007f117f42f000)
	libXau.so.6 => /usr/lib/libXau.so.6 (0x00007f117f22b000)
	libXdmcp.so.6 => /usr/lib/libXdmcp.so.6 (0x00007f117f025000)
	libbz2.so.1.0 => /usr/lib/libbz2.so.1.0 (0x00007f117f010000)
	libgpg-error.so.0 => /usr/lib/libgpg-error.so.0 (0x00007f117edf0000)

	Version information:
	./nextpnr-ice40:
		libpthread.so.0 (GLIBC_2.2.5) => /usr/lib/libpthread.so.0
		libgcc_s.so.1 (GCC_3.0) => /usr/lib/libgcc_s.so.1
		libm.so.6 (GLIBC_2.27) => /usr/lib/libm.so.6
		libm.so.6 (GLIBC_2.2.5) => /usr/lib/libm.so.6
		libc.so.6 (GLIBC_2.14) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.4) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /usr/lib/libc.so.6
		libQt5Core.so.5 (Qt_5.11) => /usr/lib/libQt5Core.so.5
		libQt5Core.so.5 (Qt_5) => /usr/lib/libQt5Core.so.5
		libQt5Gui.so.5 (Qt_5) => /usr/lib/libQt5Gui.so.5
		libstdc++.so.6 (GLIBCXX_3.4.18) => /usr/lib/libstdc++.so.6
		libstdc++.so.6 (GLIBCXX_3.4.14) => /usr/lib/libstdc++.so.6
		libstdc++.so.6 (CXXABI_1.3.1) => /usr/lib/libstdc++.so.6
		libstdc++.so.6 (CXXABI_1.3.5) => /usr/lib/libstdc++.so.6
[... 412 lines cut ...]
		libc.so.6 (GLIBC_2.4) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.3) => /usr/lib/libc.so.6
	/usr/lib/libXau.so.6:
		libc.so.6 (GLIBC_2.4) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /usr/lib/libc.so.6
	/usr/lib/libXdmcp.so.6:
		libc.so.6 (GLIBC_2.4) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /usr/lib/libc.so.6
	/usr/lib/libbz2.so.1.0:
		libc.so.6 (GLIBC_2.3) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.4) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /usr/lib/libc.so.6
	/usr/lib/libgpg-error.so.0:
		libc.so.6 (GLIBC_2.3) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.14) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.15) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.4) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.7) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.2.5) => /usr/lib/libc.so.6
		libc.so.6 (GLIBC_2.3.4) => /usr/lib/libc.so.6
```

No readline anywhere in the dependency tree as far as I can see. Did you ever investigate this further?

mmicko, 2020-01-04:
Confirmed no readline library linked anywhere in nextpnr

closed 2020-01-04 by mmicko