YosysHQ/nextpnr #84: Segfault with input driving output directly

cliffordwolf, 2018-10-03:
I get the following segfault during packing (see attached zip for test case). I think this is related to outputs being driven directly by inputs.

```
$ bash run.sh
....
Info: dbg1 feeds SB_IO qpi_in_buf[1], removing $nextpnr_obuf dbg1.
Info: dbg2 feeds SB_IO qpi_in_buf[0], removing $nextpnr_obuf dbg2.
Info: dbg3 feeds SB_IO qpi_io_buf[0], removing $nextpnr_obuf dbg3.
Info: dbg4 feeds SB_IO qpi_io_buf[1], removing $nextpnr_obuf dbg4.
Info: qpi_io2 feeds SB_IO qpi_io_buf[2], removing $nextpnr_iobuf qpi_io2.
run.sh: line 2: 10962 Segmentation fault      nextpnr-ice40 --freq 25 --up5k --asc mlaccel.asc --pcf mlaccel.pcf --json mlaccel.json
```

Running it in valgrind gives me a bunch of invalid read errors, then it exits using the following message during timing budgeting:

```
Info: Annotating ports with timing budgets for target frequency 25.00 MHz
ERROR: no timing info for port 'I' of cell type '$nextpnr_obuf'
```

This is with current git master (2298c89).

[nextpnr_segfault.zip](https://github.com/YosysHQ/nextpnr/files/2441649/nextpnr_segfault.zip)

daveshah1, 2018-10-03:
Can you provide the source Verilog as well as the JSON?

cliffordwolf, 2018-10-03:
See commit f4bfb2a in https://github.com/SymbioticEDA/mlaccel. (Sorry everyone reading along, it's not public yet. But @daveshah1 should have access.)

This is not exactly the same code as for the error above, and for me it fails with a slightly different error (bus error instead of segfault):

```
mlaccel/demo$ make mlaccel.bin 
nextpnr-ice40 --freq 25 --up5k --asc mlaccel.asc --pcf mlaccel.pcf --json mlaccel.json
...
Info: Annotating ports with timing budgets for target frequency 25.00 MHz
Makefile:58: recipe for target 'mlaccel.asc' failed
make: *** [mlaccel.asc] Bus error
```

But running it in valgrind shows "invalid read" errors at the same packing step as above.

cliffordwolf, 2018-10-03:
I don't think that code should work, btw. It just shouldn't result in segfaults and bus errors. :) I guess the problem is that the output is driven directly by an input that is connected to an SB_IO cell in the design. The output should be driven by the SB_IO D_IN_0 port in the design. I'll create a small minimal test case..

daveshah1, 2018-10-03:
Was just about to post that! Working on a better error now.

cliffordwolf, 2018-10-03:
Minimal example (drop-in for `nextpnr/ice40/blinky.v`):

```
module blinky (
    input  clki,
    output led1,
    output led2,
    output led3,
    output led4,
    output led5
);
    SB_IO #(
        .PIN_TYPE(6'b 0000_01),
        .PULLUP(1'b 1)
    ) qpi_in_buf (
        .PACKAGE_PIN(clki),
        .D_IN_0(led1)
    );

    assign led2 = clki;
    assign {led3, led4, led5} = 0;
endmodule
```

Running `blinky.sh`:

```
...
Info: Annotating ports with timing budgets for target frequency 12.00 MHz
blinky.sh: line 4: 12767 Bus error               ../nextpnr-ice40 --json blinky.json --pcf blinky.pcf --asc blinky.asc
```

PR #85 (2018-10-03): ice40: Add error for bad PACKAGE_PIN connections
Fixes #84
closed 2018-10-03 by daveshah1