YosysHQ/nextpnr #87: Error on install

ameetgohil, 2018-10-03:
I get the following message trying to install 
```
>cmake -DARCH=ecp5 -DTRELLIS_ROOT=/home/qubits/proj/prjtrellis  .
-- Found PythonInterp: /usr/bin/python3.6 (found suitable version "3.6.6", minimum required is "3.5") 
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   thread
--   program_options
--   system
--   chrono
--   date_time
--   atomic
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py366; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:124 (find_package)

-- Could NOT find Boost
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py36; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:131 (find_package)

-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   python-py36
--   filesystem
--   thread
--   program_options
--   system
--   chrono
--   date_time
--   atomic
-- Configuring architecture : ecp5
-- Configuring done
-- Generating done
-- Build files have been written to: /home/qubits/proj/nextpnr
```
```
> make -j
[  2%] Built target bbasm
[  3%] Automatic MOC for target gui_ecp5
[  5%] Automatic MOC for target QtPropertyBrowser
[  6%] Generating ecp5/chipdbs/chipdb-45k.bba
[  7%] Generating ecp5/chipdbs/chipdb-25k.bba
[  9%] Generating ecp5/chipdbs/chipdb-85k.bba
[  9%] Built target gui_ecp5_autogen
[  9%] Built target QtPropertyBrowser_autogen
[ 42%] Built target gui_ecp5
[ 57%] Built target QtPropertyBrowser
Traceback (most recent call last):
  File "/home/qubits/proj/nextpnr/ecp5/trellis_import.py", line 381, in <module>
Traceback (most recent call last):
  File "/home/qubits/proj/nextpnr/ecp5/trellis_import.py", line 381, in <module>
    main()
    main()
  File "/home/qubits/proj/nextpnr/ecp5/trellis_import.py", line 347, in main
  File "/home/qubits/proj/nextpnr/ecp5/trellis_import.py", line 347, in main
    pytrellis.load_database(database.get_db_root())
    pytrellis.load_database(database.get_db_root())
RuntimeError: /home/qubits/proj/prjtrellis/database/devices.json: cannot open file
RuntimeError: /home/qubits/proj/prjtrellis/database/devices.json: cannot open file
Traceback (most recent call last):
  File "/home/qubits/proj/nextpnr/ecp5/trellis_import.py", line 381, in <module>
    main()
  File "/home/qubits/proj/nextpnr/ecp5/trellis_import.py", line 347, in main
    pytrellis.load_database(database.get_db_root())
RuntimeError: /home/qubits/proj/prjtrellis/database/devices.json: cannot open file
CMakeFiles/ecp5_chipdb.dir/build.make:79: recipe for target 'ecp5/chipdbs/chipdb-25k.bba' failed
make[2]: *** [ecp5/chipdbs/chipdb-25k.bba] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/ecp5_chipdb.dir/build.make:84: recipe for target 'ecp5/chipdbs/chipdb-45k.bba' failed
make[2]: *** [ecp5/chipdbs/chipdb-45k.bba] Error 1
CMakeFiles/ecp5_chipdb.dir/build.make:89: recipe for target 'ecp5/chipdbs/chipdb-85k.bba' failed
make[2]: *** [ecp5/chipdbs/chipdb-85k.bba] Error 1
CMakeFiles/Makefile2:143: recipe for target 'CMakeFiles/ecp5_chipdb.dir/all' failed
make[1]: *** [CMakeFiles/ecp5_chipdb.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

daveshah1, 2018-10-03:
Did you run `./download-latest-db.sh` inside the prjtrellis folder to get the database which is needed to build the chipdb? What is inside the prjtrellis/database folder?

ameetgohil, 2018-10-03:
Everything works. Closing issue. Thanks!

closed 2018-10-03 by ameetgohil