YosysHQ/nextpnr #97: File dialogs don't work on Xubuntu 18.04

thomask77, 2018-10-26:
Hi!

I just compiled nextpnr from git on Xubuntu 18.04 running in VirtualBox.  (simple  `cmake -DARCH=ice40 .`  , `make`, `sudo make install`).

If I start `nextpnr-ice40 --gui`, and click on File-> Open, the window frame appears, but without contents. Application freezes until I close the window.

Console says:
`Gtk-Message: 21:18:59.190: GtkDialog mapped without a transient parent. This is discouraged.`

Is there anything I can do to help debugging this?

best regards,
Thomas

mmicko, 2018-10-27:
Hi,

Was not able to reproduce your problem. Is this fresh installation of 18.04, or some upgrade ? Thing is that we use QT5 for GUI functionality and not GTK directly. If you can try at least doing update/upgrade packages and then recompile nextpnr again with new and try again ?

eddiehung, 2018-11-02:
I'm actually running on a new, clean, xubnutu 18.04 build right now and running the latest master 0a52133 and while I do see the same Gtk-Message as you, I still get a file dialog from which I can select a project.

As @mmicko says, would you mind rebuilding and retrying?

thomask77, 2018-11-03:
It seems to be a VirtualBox (5.2.14 on Win10) problem..
I just tried it on both a native machine and in VMware, and everything works.

With VBox 3D acceleration disabled, I have the problem as described.

With VBox 3D accelaration enabled, nextpnr doesn't work at all:
```
thomask@vbox-xubuntu ~ 
$ nextpnr-ice40 --gui
OpenGL Warning: glXChooseVisual: bad attrib=0x186a0, ignoring
OpenGL Warning: glXChooseVisual: bad attrib=0x186a1, ignoring
OpenGL Warning: vboxCall failed with VBox status code VERR_BUFFER_OVERFLOW

OpenGL Warning: glXChooseVisual: bad attrib=0x186a0, ignoring
OpenGL Warning: glXChooseVisual: bad attrib=0x186a1, ignoring
[1]    2178 segmentation fault (core dumped)  nextpnr-ice40 --gui
```

Interestingly, nextpnr is the only application that does have these problems. GTKwave, etc. works without problems.

thomask77, 2018-11-03:
Very strange observation:
* If I zoom in 10-15 times, I can open the dialogs.
* If fully zoomed out, the file dialogs won't open.

So it's likely the high number of drawn elements triggering some problem in VirtualBox :/

eddiehung, 2018-11-03:
Thanks for trying all these combinations @thomask77 -- this is all useful information -- and I'm sorry that none of what you tried fixes it. Is it possible to use a VMWare instead of VBox then?
I'm going to rename the title of this issue to reflect these findings.

Alternatively, I believe nextpnr supports Windows natively -- calling @mmicko to confirm.

mmicko, 2018-11-03:
Right, OpenGL is not that greatly supported on VMs so there could be some issues like that.
But @eddiehung  is right Windows native support is working, and you can use VS2015/17 to build it or even mingw (following linux build process)

thomask77, 2018-11-03:
Thanks for your great support!
I think, I can live with the zoom-out workaround for now.

closed 2019-06-18 by thomask77