YosysHQ/nextpnr #98: Use of an SB_GB_IO fails with error -- works under arachne-pnr

RGD2, 2018-10-27:
rebuilding a known-to-work in arachne-pnr design, I get an error:

    ERROR: no timing info for port 'GLOBAL_BUFFER_OUTPUT' of cell type 'SB_GB_IO'

And then it stops. Is SB_GB_IO not supported yet?

eddiehung, 2018-11-09:
Thanks for the report! I believe this needs some more work on the ice40 device model. Assigning to @daveshah as he seems to know this area best...

eddiehung, 2018-11-11:
@dlharmon `SB_GB` should be supported right now; my understanding is that we just don't support the IO+GB combined block. Can you provide a testcase under a new issue? Thanks!

dlharmon, 2018-11-11:
@eddiehung I've confirmed SB_GB is fine on the HX1K. I've tried to make a minimal test case and failed. Will create an issue once #106 is resolved as I run into that one on my test case after removing the SB_GB.

smunaut, 2018-11-20:
@RGD2 Could you test and confirm your issue is fixed ?

RGD2, 2018-11-22:
Not sure if fixed but different issue, or just still broken with a different error. 
Either way, here's my test case: (6.65MB)
[nextpnr-bug#98.tar.gz](https://github.com/YosysHQ/nextpnr/files/2606368/nextpnr-bug.98.tar.gz)

RGD2, 2018-11-22:
The new error is:

    ERROR: Bel 'X16/Y0/io1' of type 'SB_IO' is not valid for cell '_mod.io[12]._io' of type 'SB_IO'
    ERROR: Placing design failed.

RGD2, 2018-11-22:
I think it's a different fault, but still same issue -- test case which arachne-pnr does, but nextpnr doesn't.  So, progress at least,

Thanks.

smunaut, 2018-11-22:
Yeh, it's a completely different issue. I'd open another ticket for that.

The fact is arachnepnr is a bit smarter at placing PLL. nextpnr just takes the first spot and that conflicts with one of your IO. (The IO site where the PLL is placed can't be used as input).

For the time being you can workaround it by just telling next-pnr where to put the PLL manually.
Add a "BEL" =  "X16/Y33/pll_3"  attribute on the PLL instance and then it works.

RGD2, 2018-11-22:
OK, can confirm, thanks!

For spectator's reference:

I found that putting the line:

    (* BEL = "X16/Y33/pll_3" *)
    
On the line immediately before the PLL instance in the .v file where that lived worked fine for this.

Completely satisfied!

closed 2018-11-22 by RGD2