YosysHQ/nextpnr #99: Specifying DYNAMICDELAY on iCE40 PLL may cause crash or errors.

tux3, 2018-10-27:
Specifying `DYNAMICDELAY` on a `SB_PLL40_CORE` will *sometimes* result in a `std::out_of_range` exception, apparently thrown by [the map in this assert](https://github.com/YosysHQ/nextpnr/blob/master/common/nextpnr.cc#L249). 

Note that this does not always happen, but seemingly depends on how the clock is actually used, how much work is done, and the current phase of the moon.

I'm a little out of my depth here, so all I have is a repro and some observations that I can't make sense of:

- Commenting out the `DYNAMICDELAY` results in a clean compile.
- Simplifying work (e.g. replacing `+1` by `+0`, or writing a 0 in `crash_data`) results in a clean compile.
- Making `crash_data` less than 4 bits wide, or `crash_counter` less than 3 results in this error instead:

>ERROR: no timing info for port 'DYNAMICDELAY[0]' of cell type 'SB_PLL40_CORE'

Here is the code and command line. I used Yosys and nextpnr master.

```verilog
// yosys -p 'synth_ice40 -top top -json crash.json' top.v && nextpnr-ice40 --lp8k --json crash.json
module top(input CLK_16MHZ, output DATA_OUT);
wire clk;
reg [4:0] crash_data;
reg [2:0] crash_counter;
assign DATA_OUT = crash_data[crash_counter];

SB_PLL40_CORE #(.FEEDBACK_PATH("SIMPLE"), .DIVR(4'b0000), .DIVF(7'b0111111), .DIVQ(3'b100)) 
pll_inst (
    .DYNAMICDELAY(8'b0),
    .REFERENCECLK(CLK_16MHZ),
    .PLLOUTCORE(clk),
);

always @(posedge clk) begin
    crash_counter <= crash_counter + 1;
    crash_data <= !clk; // By the way, shouldn't Yosys deduce clk as a constant here? Writing a 0 compiles "cleanly".
end
endmodule
```

daveshah1, 2018-10-27:
Thanks for the report! I think this should be fixed by af9ed37

tux3, 2018-10-27:
Thank you! I can confirm the fix works for my project.

closed 2018-10-27 by tux3