YosysHQ/nextpnr #106: Assertion failure: unable to find config bit ColBufCtrl.glb_netwk_0 on ICE40 LP384

dlharmon, 2018-11-11:
```
terminate called after throwing an instance of 'nextpnr_ice40::assertion_failure'
  what():  Assertion failure: unable to find config bit ColBufCtrl.glb_netwk_0 (/home/dlharmon/software/ice40/nextpnr/ice40/bitstream.cc:41)
```

I was able to reduce my project to a simple LED blinker while still getting the assertion failure. Pin number for the clock or LED doesn't seem to matter. Fortunately, I can fall back to arachne for these ICE40LP384 projects and try out nextpnr on the HX1K projects.

versions (about 12 hours old at this point)
```
dlharmon@dlhlaptop:~/nextpnr_assertion_failure_ColBufCtrl$ yosys -V
Yosys 0.8+46 (git sha1 5387ccb0, clang 3.8.1-24 -fPIC -Os)
dlharmon@dlhlaptop:~/nextpnr_assertion_failure_ColBufCtrl$ nextpnr-ice40 --version
nextpnr-ice40 -- Next Generation Place and Route (git sha1 15d9b3d)
```

top.v
```
module top(input clock,
           output led);

   reg [23:0]    count = 0;

   always @ (posedge clock)
        count <= count + 1'b1;

   assign led = count[23];

endmodule
```

constraints.pcf:
```
set_io led 18
set_io clock 19
```

Makefile
```
SOURCES = top.v

all: fine.asc

fine.json: $(SOURCES)
	yosys -q -p 'synth_ice40 -top top -json fine.json' $(SOURCES)

fine.asc: fine.json constraints.pcf
	nextpnr-ice40 --lp384 --json fine.json --pcf constraints.pcf --asc fine.asc
```

MVCE repo
=====
https://github.com/dlharmon/nextpnr_assertion_failure_ColBufCtrl

You should be able to just run make to reproduce the failure assuming nextpnr and yosys are installed.

daveshah1, 2018-11-11:
Thanks for your report and MCVE! 

Can you confirm that #111 fixes your issue (and works on hardware, if you are able to test)?

dlharmon, 2018-11-11:
This issue does appear to be resolved. One design works on hardware. The other, I may be filing another issue for (Warning: giving up after 200 iterations.). Thanks!

PR #111 (2018-11-11): ice40: Don't set colbuf bits for 384
Fixes #106
closed 2018-11-11 by daveshah1