YosysHQ/nextpnr #117: ERROR: Assert `src_wire != WireId()' failed in nextpnr/common/router1.cc:869.

smunaut, 2018-11-14:
If you have an `inout` wire on the top level that's not connected to anythng in the design, this will trigger the assert in the title when checking the routing in next-pnr.

Minimal test case : 

```
module bug (
    inout wire gpio
);

endmodule
```

whitequark, 2018-11-20:
Workaround (thanks @daveshah1):
```diff
diff --git a/common/router1.cc b/common/router1.cc
index 198461b..476b2ae 100644
--- a/common/router1.cc
+++ b/common/router1.cc
@@ -868,7 +868,8 @@ bool Context::checkRoutedDesign() const
         }
 
         auto src_wire = ctx->getNetinfoSourceWire(net_info);
-        log_assert(src_wire != WireId());
+        if (src_wire == WireId())
+            continue;
 
         if (net_info->wires.count(src_wire) == 0) {
             if (ctx->debug)
```

smunaut, 2018-11-20:
Also related, if the `inout` port is used as output only you get a different assertion failure:

`nextpnr-ice40: nextpnr/json/jsonparse.cc:631: void nextpnr_ice40::JsonParser::insert_iobuf(nextpnr_ice40::Context*, nextpnr_ice40::NetInfo*, nextpnr_ice40::PortType, const string&): Assertion `net->driver.cell->type == ctx->id("$nextpnr_iobuf")' failed.`

PR #139 (2018-11-26): router1: Fix unrouted, undriven nets
Fixes #117
closed 2018-11-26 by daveshah1