YosysHQ/nextpnr #120: ERROR: No wire found for port PACKAGE_PIN on destination cell SB_IO.

mithro, 2018-11-16:
I'm getting the following error when compiling a design;
```
Info: Routing..
ERROR: No wire found for port PACKAGE_PIN on destination cell SB_IO_1.
ERROR: Routing design failed.
```

I'm not sure what it means? I *think* the important verilog is below;

```verilog

module top(
	input clk16,
	output reg spiflash_cs_n,
	output reg spiflash_clk,
	output reg spiflash_mosi,
	input spiflash_miso,
	output spiflash_wp,
	output spiflash_hold,
	input serial_rx,
	output reg serial_tx,
	inout usb_d_p,
	inout usb_d_n,
	output usb_pullup,
	output user_led0
);

SB_IO #(
	.PIN_TYPE(6'b101001),
	.PULLUP(1'd0)
) SB_IO_1 (
	.D_OUT_0(usbdevice_usb_n_tx),
	.OUTPUT_ENABLE(usbdevice_usb_tx_en),
	.PACKAGE_PIN(usb_d_n),
	.D_IN_0(usbdevice_usb_n_rx_io)
);
```

I've also included the full test case in the following zip file: [wire-package-pin.zip](https://github.com/YosysHQ/nextpnr/files/2588001/wire-package-pin.zip)

daveshah1, 2018-11-16:
@smunaut has pointed out (on IRC) that `usb_d_p` and `usb_d_n` are connected both to `PACKAGE_PIN` of a `SB_IO`, and also logic in the following statement:

     assign mux_payload_data = {usb_d_n, usb_d_p};

Although this is not legal, nextpnr should fail more gracefully (as it does already if `PACKAGE_PIN` is connected to more than one top level IO, for example).

smunaut, 2018-11-16:
Something like : 

```diff
diff --git a/ice40/pack.cc b/ice40/pack.cc
index 43a32b3..9ca95ee 100644
--- a/ice40/pack.cc
+++ b/ice40/pack.cc
@@ -438,6 +438,11 @@ static void pack_io(Context *ctx)
             }
             packed_cells.insert(ci->name);
             std::copy(ci->attrs.begin(), ci->attrs.end(), std::inserter(sb->attrs, sb->attrs.begin()));
+        } else if (is_sb_io(ctx, ci)) {
+            NetInfo *net = ci->ports.at(ctx->id("PACKAGE_PIN")).net;
+            if ((net != nullptr) && (net->users.size() > 1))
+                log_error("PACKAGE_PIN of SB_IO '%s' connected to more than a single top level IO.\n",
+                          ci->name.c_str(ctx));
         }
     }
     for (auto pcell : packed_cells) {
```

PR #124 (2018-11-16): ice40: Add warning if an instanciated SB_IO has its PACKAGE_PIN used elsewhere
Fixes #120
closed 2018-11-16 by daveshah1