YosysHQ/nextpnr #121: Assertion failure: init != "" (/home/olof/code/icestorm/nextpnr/ice40/bitstream.cc:778)

olofk, 2018-11-16:
I'm hitting an assertion failure with cfaa6c0e5de72e25da996f1fcda1d9caa58e4d1f (have not tried other versions). Don't have the time to create a minimal example now, but I can try to provide that later if required. In the meantime, I'm attaching the synthesized project in the hope that it's something easy to figure out. `make serv_0.bin` should do the trick. Shout if you need the source tree as well
[tinyfpga_bx-icestorm.tar.gz](https://github.com/YosysHQ/nextpnr/files/2589212/tinyfpga_bx-icestorm.tar.gz)

daveshah1, 2018-11-16:
It seems that I mistakenly added that assertion for debugging, and it was only tested with Yosys inferred BRAM which always have INIT parameters on them. Can you check that it works with the assertion removed?

olofk, 2018-11-16:
That seems to work just fine. Thanks!

commit 20aa0a0e (2018-11-16): ice40: Remove unnecessary RAM assertion

Fixes #121
closed 2018-11-16 by daveshah1