YosysHQ/nextpnr #128: List boost packages to install

xobs, 2018-11-19:
The `README.md` file says to install `libboost-dev` on Ubuntu, however this is not enough.  Other Boost packages must also be installed, in addition to `libboost-dev`:

```sh
$ sudo apt install libboost-dev libboost-filesystem-dev libboost-thread-dev libboost-program-options-dev libboost-python-dev
```

eddiehung, 2018-11-20:
Thanks for posting your findings @xobs ! Would you like to create a PR for an updated README.md, and we'll merge it in?

My solution (at least on Ubuntu 18.04) to this was to install "libboost-all-dev" which I believe includes all of these too.

xobs, 2018-11-22:
It does include all of those, but the documentation indicates that just `libboost-dev` is fine.  I've submitted bfbea5bcb7a79297a598dca1408d8321cf637982 which specifies the required packages.