YosysHQ/nextpnr #129: cmake: "Could NOT find Boost -- Boost version: 1.58.0"

xobs, 2018-11-19:
When running `cmake`, a confusing and contradictory output is presented, indicating that both cmake couldn't find Boost, and that it found Boost 1.58.0, on Ubuntu 16.04:

```sh
xobs@Pazuzu:~$ cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/opt/nextpnr
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   thread
--   program_options
--   system
--   chrono
--   date_time
--   atomic
-- Could NOT find Boost
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   python-py35
--   filesystem
--   thread
--   program_options
--   system
--   chrono
--   date_time
--   atomic
-- Configuring architecture : ice40
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/Users/smcro/Documents/Code/Tomu/nextpnr
xobs@Pazuzu:~$
```

eddiehung, 2018-11-20:
Hi @xobs -- yes I can see why this could be confusing, and it looks to be an artifact of how it needs to run multiple searches for the Boost Python libraries, since different distros name them different things.

I've added a `QUIET` option to those searches so that my output (on 18.04) looks like this now:
```
eddie@eddies-desktop:~/nextpnr/master$ cmake -DARCH=ice40
-- Boost version: 1.65.1
-- Found the following Boost libraries:
--   filesystem
--   thread
--   program_options
--   system
--   chrono
--   date_time
--   atomic
CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py366; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:124 (find_package)

CMake Warning at /usr/share/cmake-3.10/Modules/FindBoost.cmake:1626 (message):
  No header defined for python-py36; skipping header check
Call Stack (most recent call first):
  CMakeLists.txt:131 (find_package)

-- Configuring architecture : ice40
-- Configuring done
-- Generating done
-- Build files have been written to: /home/eddie/nextpnr/master
```
Can you check if PR #134 behaves the same for you too?

xobs, 2018-11-22:
Thanks!  This clears up the confusion.

closed 2018-11-22 by eddiehung

flowpoint, 2019-02-16:
hit the same error on fedora:
`CMake Error at CMakeLists.txt:179 (message):
  No version of Boost::Python 3.x could be found.
`
simple workaround for me seems adding:
 `-DPYTHON_EXECUTABLE:FILEPATH=/usr/bin/python3`