YosysHQ/nextpnr #145: Crash using SB_PLL40_PAD

adamgreig, 2018-11-24:
I'm trying to get a simple PLL example using `SB_PLL40_PAD` working on an iCE40 HX8K. Arachne successfully routes the design, but nextpnr segfaults computing timing budgets.

`top.v`:
```verilog
module top (
    input CLK_OSC100,
    output LED1
);

wire clk;

(* BEL="X16/Y33/pll_3" *)
SB_PLL40_PAD #(
    .FEEDBACK_PATH("SIMPLE"),
    .DIVR(4'b0100),
    .DIVF(7'b0101111),
    .DIVQ(3'b101),
    .FILTER_RANGE(3'b010)
) pll (
    .RESETB(1'b1),
    .BYPASS(1'b0),
    .PACKAGEPIN(CLK_OSC100),
    .PLLOUTCORE(clk)
);

reg [24:0] count;
assign LED1 = count[24];

always @(posedge clk) begin
    count <= count + 1;
end

endmodule
```

The attribute on `PLL` is required to ensure it places the bottom PLL instead of the top PLL, which would otherwise error because you can't set the `PACKAGEPIN` to a pin not on that PLL's cell.

`top.pcf`:
```
set_io LED1 70
set_io CLK_OSC100 129
```

```
$ yosys -q -p "synth_ice40 -top top -json top.json" top.v
$ gdb nextpnr-ice40
(gdb) set args --hx8k --package tq144:4k --json top.json --pcf top.pcf --asc top.asc
(gdb) r
Starting program: /usr/local/bin/nextpnr-ice40 --hx8k --package tq144:4k --json top.json --pcf top.pcf --asc top.asc
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Info: Importing module top
Info: Rule checker, Verifying pre-placed design
Info: Checksum: 0xe76c5eea

Info: constrained 'LED1' to bel 'X31/Y0/io0'
Info: constrained 'CLK_OSC100' to bel 'X16/Y33/io1'

Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info: Packing non-LUT FFs..
Info: Packing carries..
Info: Packing RAMs..
Info: Packing special functions..
Info:   constrained PLL 'pll_PLL' to X16/Y33/pll_3
Info: Promoting globals..
Info: Constraining chains...
Info: Checksum: 0xc18f715f

Info: Annotating ports with timing budgets for target frequency 12.00 MHz

Program received signal SIGSEGV, Segmentation fault.
nextpnr_ice40::Arch::getPortTimingClass (this=0x555560d49470, cell=0x0, port=...,
    clockInfoCount=@0x7fffffffd200: 0) at /home/adam/opt/nextpnr/ice40/arch.cc:881
881         if (cell->type == id_ICESTORM_LC) {
(gdb) p cell
$1 = (const nextpnr_ice40::CellInfo *) 0x0
(gdb) bt
#0  nextpnr_ice40::Arch::getPortTimingClass (this=0x555560d49470, cell=0x0, port=...,
    clockInfoCount=@0x7fffffffd200: 0) at /home/adam/opt/nextpnr/ice40/arch.cc:881
#1  0x00005555557a3cea in nextpnr_ice40::Timing::walk_paths (this=this@entry=0x7fffffffd420)
    at /home/adam/opt/nextpnr/common/timing.cc:193
#2  0x000055555579d5f9 in nextpnr_ice40::Timing::assign_budget (this=0x7fffffffd420)
    at /home/adam/opt/nextpnr/common/timing.cc:469
#3  nextpnr_ice40::assign_budget (ctx=0x555560d49470, quiet=quiet@entry=false)
    at /home/adam/opt/nextpnr/common/timing.cc:481
#4  0x00005555557476ec in nextpnr_ice40::CommandHandler::executeMain (this=this@entry=0x7fffffffd980,
    ctx=std::unique_ptr<nextpnr_ice40::Context> = {...}) at /home/adam/opt/nextpnr/common/command.cc:234
#5  0x0000555555747d3e in nextpnr_ice40::CommandHandler::exec (this=0x7fffffffd980)
    at /home/adam/opt/nextpnr/common/command.cc:291
#6  0x0000555555742f04 in main (argc=<optimised out>, argv=<optimised out>)
    at /home/adam/opt/nextpnr/ice40/main.cc:161
```

Yosys and nextpnr-ice40 are both current master:
```
$ nextpnr-ice40 --version
nextpnr-ice40 -- Next Generation Place and Route (git sha1 8bda861)
$ yosys -V
Yosys 0.8+53 (git sha1 ab97edde, clang 6.0.0-1ubuntu2 -fPIC -Os)
```

daveshah1, 2018-11-24:
Thanks for reporting! Could you check that #146 fixes this issue for you?

adamgreig, 2018-11-24:
Thanks for the very quick fix! I confirm #146 fixes my issue.

PR #146 (2018-11-24): ice40: Fix disconnection of PACKAGEPIN for PAD PLLs
Fixes #145
closed 2018-11-24 by daveshah1