YosysHQ/nextpnr #151: LVDS clock input with SB_GB_IO not working

szabolor, 2018-11-26:
I'd like to use a differential ("LVDS") clock input on an iCE40HX4K-TQ144 device. I haven't found any example code regarding input LVDS clock signals in nextpnr repository, so I could only rely on the description and example in "TN1253 Using Differential I/O (LVDS, Sub-LVDS) in iCE40 LP/HX Devices". Based on that, here is a minimal example for my use case:
```
module top (
  input wire in_n,
  output wire out
);

wire clk;

SB_GB_IO #(
  .PIN_TYPE(6'b000000),
  .IO_STANDARD("SB_LVDS_INPUT")
) differential_clock_input (
  .PACKAGE_PIN(in_n),
  .GLOBAL_BUFFER_OUTPUT(clk)
);

assign out = cntr;
reg cntr = 1'd0;

always @(posedge clk) begin
  cntr <= !cntr;
end

endmodule
```
pins.pcf: 
```
set_io in_n 20
set_io out 1
```
Synthesizing the code above with yosys and then using nextpnr create the binary without any error, but the design doesn't work with an LVDS clock signal (in this case there is no change on `out`). Doing the P&R phase with arachne-pnr makes it working! Also with nextpnr, when I change `SB_GB_IO` to `SB_IO` and use D_IN_0 as a clock signal, it seems to be working. I believe the problem is connected to the `SB_GB_IO` implementation in nextpnr.
I'm not sure it is related, but doing a cross-check with different tools, from the code above icecube2 and arachne-pnr generate this as the LVDS clock input iotile:
```
.io_tile 0 17
IoCtrl LVDS
IoCtrl REN_0
IoCtrl REN_1
```
but nextpnr does this:
```
.io_tile 0 17
IoCtrl IE_0
IoCtrl IE_1
IoCtrl LVDS
IoCtrl REN_0
IoCtrl REN_1
```

I'm quite novice with these tools, maybe I've overlooked something... Can you help me how should I use an LVDS clock input signal in nextpnr?

smunaut, 2018-11-27:
Mmm, yeah, I see what's wrong.

I think several things should be changed in bitstream.cc
 - Use the ioInfo.lvds instead of re-testing the IO_STANDARD
 - Currently the is_lvds flag just changes the input_en and pullup value (but then get overwritten by the pad and global buffer stuff). I think it shouldn't change the input_en flag, but just configure the IO differently if input_en and lvds flags are set. After all if input_en is false, the fact it's LVDS is not relevant and the IO site should have the input path disabled anyway.
 - When configuring unused IOs we actually need to check not just the is_lvds flag but also if the complement input was really enabled or not.

smunaut, 2018-11-27:
You can try out this commit : https://github.com/smunaut/nextpnr/commit/bb4dacf3dbca909426f6263fd8bbab6985291830

Seems to work (comparing .asc). But I haven't had the time to validate it against other cases to make sure it doesn't break anything else.

smunaut, 2018-11-28:
This has been merged, so just try master to confirm it fixes your issue on real hardware and if it does, close the ticket.

szabolor, 2018-11-28:
I've tested my design, and now the LVDS clock input works indeed! Thank you again for your patch!

closed 2018-11-28 by szabolor