YosysHQ/nextpnr #167: Mouse coordinates broken in floorplan view on HiDPI displays

whitequark, 2018-12-05:
See the screenshot. The popup is positioned correctly, the highlighted wire/cell uses ½ of the actual coordinate.
[image]

(Also, everything appears as if it is a raster upscaled 2x; blurry.)

mmicko, 2018-12-05:
Can you please try adding in common/command before Application
initialization :

QGuiApplication::setAttribute(Qt::AA_EnableHighDpiScaling)

If it is fine please note that patch should check for Qt5.6 sine this not
available before and we support older version since it is delivered with
Ubuntu 16.04

whitequark, 2018-12-05:
Absolutely nothing changed.

mmicko, 2018-12-05:
Sorry on phone now, but it seams it should be :
QGuiApplication::setAttribute(Qt::AA_DisableHighDpiScaling)

whitequark, 2018-12-05:
That worked, but the text in the popup is way too small now. It needs to be scaled up.

whitequark, 2018-12-05:
Oh, and the icons on the toolbar are really tiny now.

q3k, 2018-12-05:
Disabling HighDPIScaling is something we should absolutely not do.

The likely culprit is the [hacky reverse projection code](https://github.com/YosysHQ/nextpnr/blob/master/gui/fpgaviewwidget.cc#L741-L769) we're currently using. I don't have a HiDPI setup to debug this - I can personally only attempt to rewrite this to be less terrible and easier to reason about.

whitequark, 2018-12-06:
@q3k You can use `QT_SCREEN_SCALE_FACTORS=2 nextpnr-ice40 --gui`, you do not need an actual HiDPI display.

q3k, 2018-12-06:
@justMaku just tested this for me and says the following should fix it: 

    -    const qreal retinaScale = devicePixelRatio();
    +    const qreal retinaScale = 1; // devicePixelRatio();

In https://github.com/YosysHQ/nextpnr/blob/master/gui/fpgaviewwidget.cc#L745. Could you try that out?

justMaku, 2018-12-07:
After doing some digging, this is basically a result of 3 separate bugs.

* **Wrong popup position**
Qt5 seems to handle high DPI displays automatically so the multiplication by `devicePixelRatio()` in https://github.com/YosysHQ/nextpnr/blob/master/gui/fpgaviewwidget.cc#L745 is not needed.

* Wrong popup scaling
The popup drawing is handled by ImGUI used with the qtimgui wrapper, which set's up the correct scaling factor in https://github.com/seanchas116/qtimgui/blob/master/ImGuiRenderer.cpp#L276 but for some reason it's ignored or not implemented properly inside of the ImGUI itself. Setting that scaling factor to `(1, 1)` seems to fix the blurriness, but then the popup itself is obviously smaller on the high DPI displays and it's position is wrong. This can be fixed by multiplying all of the coordinates and sizes by `devicePixelRatio()` in https://github.com/YosysHQ/nextpnr/blob/master/gui/fpgaviewwidget.cc#L338

This obviously is not the correct solution though, and more extensive research is needed. 

* Upscaled icons in toolbar

I didn't look closely into it but I assume Qt has a mechanism for drawing higher definition assets that should be used here, ideally with just some vector icons instead.

jay20162016, 2021-05-09:
Is this bug going to be addressed? I can manually fix the https://github.com/YosysHQ/nextpnr/blob/master/gui/fpgaviewwidget.cc#L745 part, but that requires building from source, which would not be convenient for everyone.

commit 473723b2 (2021-05-31): Fix hidpi, fixes #167, fixes #275, fixes #425
closed 2021-05-31 by mmicko