YosysHQ/nextpnr #170: terminate called after throwing an instance of 'std::invalid_argument' what(): stoi

RGD2, 2018-12-06:
```
terminate called after throwing an instance of 'std::invalid_argument'
  what():  stoi
./breaks_nextpnr.sh: line 2: 32876 Aborted                 nextpnr-ice40 --hx8k --package ct256 --freq 80 --asc hx8kbb.asc --pcf hx8kbb.pcf --json hx8kbb.json
```

RGD2, 2018-12-06:
[nextpnr-#170.tar.gz](https://github.com/YosysHQ/nextpnr/files/2651872/nextpnr-.170.tar.gz)

This test case reproduced with nextpnr (git rev-parse --short) a631583

RGD2, 2018-12-06:
Just tried with -r, same result.

daveshah1, 2018-12-06:
e7fc42a provides a clearer insight as to what is going on, which is that nextpnr is erroneously expecting the FEEDBACK_PATH parameter of the PLL to be numeric. I think this should be dealt with during packing.

daveshah1, 2018-12-06:
Further error reporting added in 1443636. Are you sure `NON_SIMPLE` is a valid `FEEDBACK_PATH` value? I can't find any evidence of it being supported in arachne-pnr, or in any vendor documentation.

RGD2, 2018-12-06:
the PLL in this design uses a file generated by icepll directly, so I
imagine it is correct.

RGD2, 2018-12-07:
So it is, I'm closing this issue and considering opening one against icestorm.
Thanks!

RGD2, 2018-12-07:
(for those watching, changed to 'SIMPLE' with appropriate settings was the fix).

closed 2018-12-07 by RGD2