YosysHQ/nextpnr #173: Error evaluating generator expression: $<TARGET_OBJECTS:ice40_chipdb>

uXeBoy, 2018-12-08:
Getting these errors when attempting: `cmake -DARCH=ice40 .`

Help?

```
CMake Error at ice40/family.cmake:81 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:233 (include)

CMake Error at ice40/family.cmake:81 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:233 (include)

CMake Error at ice40/family.cmake:81 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:233 (include)

CMake Error at ice40/family.cmake:81 (target_sources):
  Error evaluating generator expression:

    $<TARGET_OBJECTS:ice40_chipdb>

  The evaluation of the TARGET_OBJECTS generator expression is only suitable
  for consumption by CMake.  It is not suitable for writing out elsewhere.
Call Stack (most recent call first):
  CMakeLists.txt:233 (include)
```

daveshah1, 2018-12-08:
Thanks for the bug report! Can you please post your operating system and CMake version?

uXeBoy, 2018-12-08:
This is on Ubuntu 14.04 with CMake version 3.3.2

uXeBoy, 2018-12-09:
OK, updating to CMake 3.7.2 using [this PPA](https://launchpad.net/~adrozdoff/+archive/ubuntu/cmake) seems to have resolved this...

closed 2019-02-05 by daveshah1