YosysHQ/nextpnr #176: Assert and errors when using inout

tux3, 2018-12-12:
I have an ICE40 design that uses `inout`s to drive a Quad SPI, Lattice with Synplify and Active-HDL seem to work but nextpnr **crashes** or returns errors.
While I was trying to isolate the issue nextpnr and arachne-pnr hit several different failure modes, while the Lattice tools happily synthetized whatever garbage I threw at them :smile: 

I'm really surprised by how depending on the test case connecting a probe output can make an invalid design valid for nextpnr, a valid design invalid, or change nothing at all.

I'm using nextpnr and arachne-pnr built from today's master, and a pretty recent yosys build.
The target is the an ICE40-LP8K CM81.

## Summary

Here's a summary of the results, some of the examples are a little nonsensical but ideally nextpnr would try to synthetize something, at least for the non-circular cases:

TEST CASE           |NEXTPNR     |ARACHNE-PNR     |LATTICE/SYNPLIFY|
-|-|-|-|
Const inout         |OK          |FAIL            |OK         |
Const+clk assign    |**ASSERT**      |FAIL      |      OK|
Const+assign        |FAIL|        FAIL|            OK|
Dyn inout           |OK|          OK|              OK|
Dyn+clk assign      |FAIL|        FAIL|            OK|
Dyn+assign          |OK|          FAIL|            OK|
Circular inout      |FAIL|        FAIL    |        OK with warning|
Circular+assign     |OK|          FAIL|            OK with warning|

## Test case variations

The base code is pretty simple, and the test cases in the table above are just slight variations:

```verilog
module top(
    input CLK,
    output reg IGNORE_ME,

	inout INOUT,
    output reg PROBE
);

always @(posedge CLK)
    IGNORE_ME <= ~IGNORE_ME; // Pretend to do work (silence warning).

// <<< Variations go here! >>>

endmodule

```

#### Const inout

```verilog
assign INOUT = 'b1;
```

#### Const+clk assign

```verilog
assign INOUT = 'b1;

always @(posedge CLK)
    PROBE <= INOUT;
```

>$ make
nextpnr-ice40 --lp8k --package cm81 --json ./build/inout_repro.json --pcf ./src/constraints/pins.pcf --freq 32 --asc ./build/inout_repro.asc --seed 0
Info: Importing module top
nextpnr-ice40: ../json/jsonparse.cc:630: void nextpnr_ice40::JsonParser::insert_iobuf(nextpnr_ice40::Context*, nextpnr_ice40::NetInfo*, nextpnr_ice40::PortType, const string&): Assertion `net->driver.cell->type == ctx->id("$nextpnr_iobuf")' failed.
make: *** [Makefile:26: build/inout_repro.asc] Aborted

#### Const+assign

```verilog
assign INOUT = 'b1;

assign PROBE = INOUT;
```

Note that the probe output makes the previously valid design invalid.

#### Dyn inout

```verilog
assign INOUT = CLK ? 'bz : 'b0;
```

#### Dyn+clk assign

```verilog
assign INOUT = CLK ? 'bz : 'b0;

always @(posedge CLK)
    PROBE <= INOUT;
```

#### Dyn+assign

```verilog
assign INOUT = CLK ? 'bz : 'b0;

assign PROBE = INOUT;
```

This is the only test case where the probe output has no effect compared to no probe output.

#### Circular inout

```verilog
assign INOUT = INOUT ? 'bz : 'b0;
```

#### Circular+assign

```verilog
assign INOUT = INOUT ? 'bz : 'b0;

assign PROBE = INOUT;
```

Note that the probe output makes the previously invalid design valid.

---

Sorry for the giant wall of text!

I think at least the assert and some of the errors above must be bugs in nextpnr, but my verilog is pretty bad so the code might also be horribly broken.

Thank you!

daveshah1, 2018-12-12:
Thank you very much for such a detailed report!

I think all the designs failing in nextpnr should be fixed in YosysHQ/yosys#735. If you could double check that; that would be very helpful.

daveshah1, 2018-12-12:
In addition 9d852f0 adds a nicer error than an assert failure for a broken netlist (input/driver conflict) in the Const+clk assign case. This case should also be fixed properly in the Yosys PR too, but this might help debugging other problems in the future too.

tux3, 2018-12-12:
I can confirm that all the test cases pass on my end and my original design now synthetizes, thank you!

There seem to be a couple issues mentioning inouts in Yosys, could your fix be related to YosysHQ/yosys#665 and YosysHQ/yosys#666 ?

daveshah1, 2018-12-13:
It looks like #665 is indeed also solved; #666 still fails in arachne-pnr but the same netlist works in nextpnr so I'm assuming is an arachne issue.

closed 2018-12-13 by daveshah1