YosysHQ/nextpnr #177: Windows: nextpnr-ice40 exits with "no current thread"

xobs, 2018-12-13:
When trying to use `nextpnr-ice40` on Windows, every non-trivial operationfails with the same error.

For example, using the "blinky" demo from `ice40/`:

```powershell
[4:53:58 PM] C:/Code/nextpnr/ice40> ..\Release\nextpnr-ice40 --hx1k --json blinky.json --pcf blinky.pcf --asc blinky.asc
Info: Importing module blinky
Info: Rule checker, verifying imported design
Info: Checksum: 0x46c66898

Info: constrained 'led1' to bel 'X13/Y12/io1'
Info: constrained 'led2' to bel 'X13/Y12/io0'
Info: constrained 'led3' to bel 'X13/Y11/io1'
Info: constrained 'led4' to bel 'X13/Y11/io0'
Info: constrained 'led5' to bel 'X13/Y9/io1'
Info: constrained 'clki' to bel 'X0/Y8/io1'
Fatal Python error: PyThreadState_Get: no current thread

[4:54:09 PM] C:/Code/nextpnr/ice40>
```

There is not much information on how to build nextpnr on Windows, so I did the following:

```powershell
vcpkg install boost-filesystem:x64-windows boost-program-options:x64-windows boost-thread:x64-windows boost-python:x64-windows qt5-base:x64-windows
vcpkg integrate install
cmake -DARCH=ice40 -DICEBOX_ROOT=C:/ProgramData/icestorm/share/icebox -DCMAKE_TOOLCHAIN_FILE=C:/Code/vcpkg/scripts/buildsystems/vcpkg.cmake -DVCPKG_TARGET_TRIPLET=x64-windows -G "Visual Studio 15 2017 Win64" .
cmake --build . --config Release
```

xobs, 2018-12-13:
More information:

```powershell
[4:57:34 PM] C:/Code/nextpnr/ice40> ..\Release\nextpnr-ice40 -V
nextpnr-ice40 -- Next Generation Place and Route (git sha1 9d852f0)
[4:57:40 PM] C:/Code/nextpnr/ice40>
```

mmicko, 2018-12-13:
Hmm, I have crash only on Debug version and reason is that boost_python36-vc140-mt-gd-x64-1_68.dll is using python36.dll and not it's debug version, and that issue comes fact how Boost-python was compiled, when using Release all works as expected.

Reason why it could fail in Release in your case could be that python36.dll is loaded from other path, so to make sure just do run: set PATH=C:\Code\nextpnr\Release
That  will make sure that only files from Release folder are taken when running nextpnr.
Please send info if this fix your problem.

xobs, 2018-12-13:
If I do that, nextpnr-ice40.exe does nothing at all:

```cmd
C:\Code\nextpnr\Release>set PATH=C:\Code\nextpnr\Release

C:\Code\nextpnr\Release>nextpnr-ice40.exe -V

C:\Code\nextpnr\Release>nextpnr-ice40.exe --gui

C:\Code\nextpnr\Release>echo %errorlevel%
-1073741515

C:\Code\nextpnr\Release>
```

If I add my Python into the mix (`PATH=C:\Python37;C:\Code\nextpnr\Release`) then I get back the old behavior.

xobs, 2018-12-13:
I see.  I use Python from Chocolatey (choco install python), and they don't provide Python36.  I assumed that vcpkg would install a version of Python that it was compatible with, because the README.md notes that vcpkg will install all dependencies, however that doesn't seem to be the case.  I still see "/IC:\Python37\include" in the command line arguments getting passed to the compiler.

How do I tell cmake to use C:\Code\vcpkg\packages\python3_x64-windows\ (which is the vcpkg version of python3) and not C:\Python37 (which isn't even in my path, I don't know how it's finding that)?

mmicko, 2018-12-13:
Sorry not familiar with Chocolatey. What I did is I have downloaded version 3.6 https://www.python.org/downloads/  and then compiled vcpkg packages.
If you plan to use only command line you can add -DBUILD_GUI=OFF -DBUILD_PYTHON=OFF in order to compile it without python and qt support (GUI without python is not available as option). That would work for sure, since no python will be used.

xobs, 2018-12-13:
I tried installing python-3.6.4 from that link, but it refuses to install because I already have Python 3.7 installed.  I use Python 3.7 for other things, so I'd really rather not uninstall it from the entire system just for one package.  Are there alternative options available?

The documentation mentions that vcpkg is used for dependency installation.  Am I correct in assuming that Python is not one of the dependencies it installs, and that "python3:x64-windows" is not actually python3?

mmicko, 2018-12-13:
vcpkg compiles all packages for visual studio builds, and one of the things it compiles is python3.6  to get library so it can link to it. So it is not full python build that you can use. Sorry no alternative solution for now.

xobs, 2018-12-13:
Okay, so I guess the actual bug is that it's not possible to build nextpnr without uninstalling your normal Python and installing Python 3.6.4.

xobs, 2018-12-13:
Would it be possible to document some of the things I've discovered?  Much of it was hard to figure out, and was not very obvious:

* The `-G` option is a string that must match visual studio
* `-DCMAKE_TOOLCHAIN_FILE` which comes from the output of `vcpkg integrate install`
* `-DVCPKG_TARGET_TRIPLET` seems to be something that you just have to know (and isn't actually a triplet)
* The odd mix of `-DPYTHON_EXECUTABLE` pointing to C:\Python36 and `-DPYTHON_LIBRARY`/`-DPYTHON_INCLUDE_DIR`, neither of which are documented, and both of which point to different installations
* `-DICEBOX_ROOT`, which I think is documented now
* Needing to download the Python 3.6 zipfile and manually extract it
* The existence `cmake --build . --config Release` command, which saves having to open up the GUI

Would these go into their own "Windows" section?

mmicko, 2018-12-13:
Thanks for suggestion I will take care of it.

xobs, 2019-01-15:
Adding on to this -- I just reinstalled my machine so I had to go through the steps again.

The version of Python from python.org is broken.  It doesn't embed properly:

```
[3:42:55 PM] ~/Code/nextpnr> .\Release\nextpnr-ice40.exe --gui
Fatal Python error: Py_Initialize: unable to load the file system codec
ModuleNotFoundError: No module named 'encodings'

Current thread 0x00004844 (most recent call first):
[3:43:09 PM] ~/Code/nextpnr>
```

This is mentioned on a [Stack Overflow issue](https://stackoverflow.com/questions/5694706/py-initialize-fails-unable-to-load-the-file-system-codec), but no solution is given.

The problem is that the interpreter works, but not when you embed it.  The interpreter knows about `python36.zip`, but not the embedded version.

The fix is to create a directory beside `python.exe` called `Lib`, and extract the contents of `python36.zip` into that directory.

xobs, 2019-04-06:
With the latest README.md file, this has been documented, and this issue is now resolved.

closed 2019-04-06 by xobs