YosysHQ/nextpnr #179: Infinite loop in router1() for some seeds

tux3, 2018-12-13:
For some ~0.3% of seeds nextpnr can get stuck in an infinite loop with a handful of arcs remaining.
The output looks like this:

```
[...]
Info: Routing..
Info: Setting up routing queue.
Info: Routing 1171 arcs.
Info:            |   (re-)routed arcs  |   delta    | remaining
Info:    IterCnt |  w/ripup   wo/ripup |  w/r  wo/r |      arcs
Info:       1000 |       95        904 |   95   904 |       291
Info:       2000 |      801       1198 |  706   294 |         5
Info:       3000 |     1801       1198 | 1000     0 |         5
Info:       4000 |     2801       1198 | 1000     0 |         5
Info:       5000 |     3801       1198 | 1000     0 |         5
Info:       6000 |     4801       1198 | 1000     0 |         5
Info:       7000 |     5801       1198 | 1000     0 |         5
```

GDB says the code is stuck in `nextpnr_ice40::router1 at nextpnr/common/router1.cc:791`

I've stumbled on this bug a handful of times in the last few weeks. On the current master it's pretty much only an issue when using a script that brute-force searches for the best Fmax.

With the right input and seed this is 100% reproducible, but it's still pretty annoying to isolate since any change to the design or command line parameters shuffles the bad seed space.

The simplest repro I have to hang nextpnr [master](https://github.com/YosysHQ/nextpnr/commit/19cffde375fe63e36bfd880727bdc0a04a2ccf32) is this [150kB json input](https://gist.github.com/tux3/36726c9a54c3a18b1986be51c574f33c) and cmdline:

`nextpnr-ice40 --lp8k --package cm81 --freq 32 --seed 557 --json repro.min.json`

I'm also happy to provide the bruteforce script if that helps.

daveshah1, 2018-12-13:
Would you be able to rerun with `--verbose --debug`? This might provide a bit more information about where it is getting stuck.

tux3, 2018-12-13:
Sure, here's the log output: https://gist.githubusercontent.com/tux3/9f37cd7e232b5b21ae2e5e6d50fcddcf/raw/3a60b243503afadd4b95f9ddedc58e10ad02049a/gistfile1.txt

daveshah1, 2018-12-13:
Thanks! The problem seems to be the `$PACKER_VCC_NET` (which is relatively high-fanout). The problem LUTFF only has 6/8 LUTs used so it's not a case of running out of local tracks, but does seem to be a problem in the router (cc @cliffordwolf).

Although not a fix for the underlying problem, we should also be more intelligent and rewrite LUT functions to remove tied-high inputs (disconnected LUT inputs are 0).