YosysHQ/nextpnr #186: using pll lock causes assert error in nextpnr

janrinze, 2018-12-20:
When using a lock signal from the pll module (created with icepll) triggers an assert error.

to replicate:
untar attached file in an empty directory,

yosys -q -p 'synth_ice40 -top top -json test.json' test.v
nextpnr-ice40 --pcf hx8k.pcf --json test.json --asc test.asc --hx8k

error given is:

Info: Importing module top
Info: Rule checker, verifying imported design
Info: Checksum: 0x00b23127

Info: constrained 'led1' to bel 'X17/Y33/io0'
Info: constrained 'led2' to bel 'X16/Y33/io1'
Info: constrained 'led3' to bel 'X17/Y0/io0'
Info: constrained 'pclk' to bel 'X16/Y0/io1'

Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info: Packing non-LUT FFs..
Info: Packing carries..
Info: Packing RAMs..
Info: Placing PLLs..
Info:   constrained PLL 'mypll.uut' to X16/Y33/pll_3
Info:   (given its connections, this PLL could have been a PAD PLL)
Info: Packing special functions..
Info:   PLL 'mypll.uut' has LOCK output, need to pass all outputs via LUT
Info:   constrained '$abc$231$auto$blifparse.cc:492:parse_blif$232_LC' to X1/Y32/lc0
Info:   constrained '$auto$alumacc.cc:474:replace_alu$8.slice[15].adder_LC' to X1/Y32/lc1
Info:   constrained '$auto$alumacc.cc:474:replace_alu$8.slice[16].adder_LC' to X1/Y32/lc2
Info:   constrained '$auto$alumacc.cc:474:replace_alu$8.slice[17].adder_LC' to X1/Y32/lc3
Info:   constrained '$auto$alumacc.cc:474:replace_alu$8.slice[18].adder_LC' to X1/Y32/lc4
Info:   constrained '$auto$alumacc.cc:474:replace_alu$8.slice[19].adder_LC' to X1/Y32/lc5
Info:   constrained '$auto$alumacc.cc:474:replace_alu$8.slice[20].adder_LC' to X1/Y32/lc6
Info:   constrained '$auto$alumacc.cc:474:replace_alu$8.slice[0].adder_LC' to X1/Y32/lc7
terminate called after throwing an instance of 'nextpnr_ice40::assertion_failure'
  what():  Assertion failure: z < 8 (/home/janrinze/src/nextpnr/ice40/pack.cc:1224)
Aborted

[nextpnrtest.tar.gz](https://github.com/YosysHQ/nextpnr/files/2700018/nextpnrtest.tar.gz)

daveshah1, 2018-12-20:
Thank you for your detailed report and for including a useful test case! Can you confirm that #187 fixes your issue and, if possible, test that it works on hardware?

janrinze, 2018-12-20:
Apart from a silly flaw in the verilog file ( "if (lock)" should be "if (!lock)" and the pll being too fast to see the blinking leds) the error is gone and upload works as expected.
Thanks for the quick fix.
Next bug report is coming up. ;-)

PR #187 (2018-12-20): ice40: Fix LOCK feedthrough insertion with carry or >8 LUTs
Fixes #186
closed 2018-12-20 by daveshah1