YosysHQ/nextpnr #190: Trying to build in macOS 10.14.2 and I got this error after runnintg++g

enKoding, 2018-12-21:
4 warnings generated.
[ 51%] Linking CXX static library libQtPropertyBrowser.a
[ 51%] Built target QtPropertyBrowser
[ 52%] Linking CXX executable bbasm
[ 52%] Built target bbasm
make[2]: *** No rule to make target `/usr/local/bin/icebox/chipdb-8k.txt', needed by `ice40/chipdbs/chipdb-8k.bba'.  Stop.
make[1]: *** [CMakeFiles/ice40_chipdb.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 54%] Linking CXX static library libgui_ice40.a
[ 54%] Built target gui_ice40
make: *** [all] Error 2

daveshah1, 2018-12-21:
Set ICEBOX_ROOT to `/usr/local/share/icebox`

mmicko, 2018-12-21:
You need to git clone and install  https://github.com/cliffordwolf/icestorm in order to have chip database files for ice40

enKoding, 2018-12-21:
Worked like a charm... You rock man !

daveshah1, 2018-12-21:
Clarification added in 8f5d500.

closed 2018-12-21 by daveshah1