YosysHQ/nextpnr #191: nextpnr-ecp5 throws std::out_of_range when using inout

kbeckmann, 2018-12-22:
Getting the following callstack when building the testcase below.
```
#0  0x00007fba2ec40d7f in raise () from /usr/lib/libc.so.6
#1  0x00007fba2ec2b672 in abort () from /usr/lib/libc.so.6
#2  0x00007fba2eff758e in __gnu_cxx::__verbose_terminate_handler () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#3  0x00007fba2effddfa in __cxxabiv1::__terminate (handler=<optimized out>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:47
#4  0x00007fba2effde57 in std::terminate () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:57
#5  0x00007fba2effe0ac in __cxxabiv1::__cxa_throw (obj=obj@entry=0x55c86d43c1f0, tinfo=0x7fba2f0f0038 <typeinfo for std::out_of_range>, 
    dest=0x7fba2f013ba0 <std::out_of_range::~out_of_range() at /build/gcc/src/gcc/libstdc++-v3/src/c++98/stdexcept.cc:65>)
    at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
#6  0x00007fba2eff9781 in std::__throw_out_of_range (__s=0x55c8651ae7d0 "_Map_base::at") from /usr/lib/libstdc++.so.6
#7  0x000055c864f75b73 in nextpnr_ecp5::replace_port(nextpnr_ecp5::CellInfo*, nextpnr_ecp5::IdString, nextpnr_ecp5::CellInfo*, nextpnr_ecp5::IdString) ()
#8  0x000055c86506c17a in nextpnr_ecp5::nxio_to_tr(nextpnr_ecp5::Context*, nextpnr_ecp5::CellInfo*, nextpnr_ecp5::CellInfo*, std::vector<std::unique_ptr<nextpnr_ecp5::CellInfo, std::default_delete<nextpnr_ecp5::CellInfo> >, std::allocator<std::unique_ptr<nextpnr_ecp5::CellInfo, std::default_delete<nextpnr_ecp5::CellInfo> > > >&, std::unordered_set<nextpnr_ecp5::IdString, std::hash<nextpnr_ecp5::IdString>, std::equal_to<nextpnr_ecp5::IdString>, std::allocator<nextpnr_ecp5::IdString> >&) ()
#9  0x000055c86508fb23 in nextpnr_ecp5::Ecp5Packer::pack_io() ()
#10 0x000055c865082f08 in nextpnr_ecp5::Arch::pack() ()
#11 0x000055c864f69cb0 in nextpnr_ecp5::CommandHandler::executeMain(std::unique_ptr<nextpnr_ecp5::Context, std::default_delete<nextpnr_ecp5::Context> >) ()
#12 0x000055c864f6a258 in nextpnr_ecp5::CommandHandler::exec() ()
#13 0x000055c864f64486 in main ()
```

Minimal test case (getting the same exception if `oe` and `a` are connected to pins in the lpf as well):

testcase.v:
```
`default_nettype none

module test_inout(
	inout wire io_pin,
);
	reg a = 1'b0;
	reg oe = 0;
	assign io_pin = oe ? a : 1'bz;
endmodule
```

testcase.lpf:
```
LOCATE COMP "io_pin" SITE "G3" ;
```

Built it like this:
```
yosys -p "synth_ecp5 -json testcase.json" testcase.v
nextpnr-ecp5 --json testcase.json --basecfg $TRELLIS/misc/basecfgs/empty_lfe5u-25f.config --textcfg testcase_out.config --25k --package CABGA381 --lpf testcase.lpf
```

Tested this with the latest nextpnr (8f5d5003e99db83c87403f2f26a187e01b0e9bb0), and yosys (ee8a7589e09efeb38cbc5ee6fd241758d6637fe2).

daveshah1, 2018-12-22:
Thanks for the report and for including a useful test case! Fixed in e76479f.

closed 2018-12-22 by daveshah1

kbeckmann, 2018-12-22:
Thanks for fixing it so quickly!

commit e76479f3 (2018-12-22): ecp5: Fix tristate IO insertion

Fixes #191