YosysHQ/nextpnr #194: nextpnr-ecp5: std::out_of_range assert when writing to a pin with a BIDIR instance

kbeckmann, 2018-12-25:
Sorry for the confusing title. I found this assert when experimenting with TRELLIS_IO BIDIR instances and doing incorrect things. I am creating a TRELLIS_IO BIDIR instance for the pin pin_IO, but I am simultaneously setting pin_IO in my state machine (which is wrong). But since this asserts, I'm creating the issue. This should probably generate a fatal error instead.

```
module top(
	input clk,
	inout pin_IO,
);

wire pin_I;
reg pin_O = 16'd0;
reg pin_OE = 1'd0;
reg state = 1'd0;
reg next_state;

always @(*) begin
	pin_IO <= 16'd0;
	next_state <= state;
	case (state)
		1'd1: begin
			next_state <= 1'd0;
		end
		default: begin
			pin_IO <= 1'd1;
			next_state <= 1'd1;
		end
	endcase
end

always @(posedge clk) begin
	state <= next_state;
end

TRELLIS_IO #(
	.DIR("BIDIR")
) TRELLIS_IO (
	.I(pin_O),
	.T(pin_OE),
	.B(pin_IO),
	.O(pin_I)
);

endmodule
```

backtrace
```
#6  0x00007f84c4a0e781 in std::__throw_out_of_range (
    __s=0x55d27c5e67d0 "_Map_base::at") from /usr/lib/libstdc++.so.6
#7  0x000055d27c4bd3c4 in std::__detail::_Map_base<nextpnr_ecp5::IdString, std::pair<nextpnr_ecp5::IdString const, std::unique_ptr<nextpnr_ecp5::CellInfo, std::default_delete<nextpnr_ecp5::CellInfo> > >, std::allocator<std::pair<nextpnr_ecp5::IdString const, std::unique_ptr<nextpnr_ecp5::CellInfo, std::default_delete<nextpnr_ecp5::CellInfo> > > >, std::__detail::_Select1st, std::equal_to<nextpnr_ecp5::IdString>, std::hash<nextpnr_ecp5::IdString>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true>, true>::at(nextpnr_ecp5::IdString const&) ()
#8  0x000055d27c4bb35b in nextpnr_ecp5::Arch::pack() ()
#9  0x000055d27c3a1cb0 in nextpnr_ecp5::CommandHandler::executeMain(std::unique_ptr<nextpnr_ecp5::Context, std::default_delete<nextpnr_ecp5::Context> >) ()
#10 0x000055d27c3a2258 in nextpnr_ecp5::CommandHandler::exec() ()
#11 0x000055d27c39c486 in main ()
```

Built with nextpnr e76479f and latest yosys as of writing (245724a).

daveshah1, 2018-12-25:
See #195

kbeckmann, 2018-12-25:
Awesome! Tested the patch and it seems to work as intended (i.e. exit with a fatal error).

PR #195 (2018-12-25): ecp5: Check for incorrect use of TRELLIS_IO 'B' pin
Fixes #194
closed 2018-12-25 by daveshah1