YosysHQ/nextpnr #200: cmake -Darch=ecp5 fails

BracketMaster, 2018-12-29:
When targeting ecp5, I get an error python complains it cannot import database in the file "./ecp5/trellis_import.py ".
To fix this I do, I add to trellis_import.py
```
import sys
sys.path.append('/path/to/prjtrellis')
sys.path.append('/path/to/prjtrellis/timing/util')
sys.path.append('/path/to/prjtrellis/util/common') ```
Can someone fix the CMakeLists to add those to path upon configuring?

daveshah1, 2018-12-29:
These paths should be added automatically so long as you add `-DTRELLIS_ROOT=/path/to/prjtrellis` to your CMake command line.

BracketMaster, 2018-12-29:
Hmm... I tried on linux and Mac correctly specifying ```-DTRELLIS_ROOT=/path/to/prjtrellis``` and still no luck. @daveshah1 have you tried building the latest commit on your platform. I assuming you're on Linux?

closed 2019-07-29 by BracketMaster