YosysHQ/nextpnr #209: LVDS clock input results in assertion failure

bgamari, 2019-01-12:
```verilog
// test.v
module test (input  PIN_5, output wire PIN_18);
   lvds_clock_input in (PIN_5, PIN_18);
endmodule

module lvds_clock_input ( input pin, output wire rd );
   SB_GB_IO #(.PIN_TYPE(6'b0000_00),
           .IO_STANDARD("SB_LVDS_INPUT")) lvds_in
     (.PACKAGE_PIN(pin), .GLOBAL_BUFFER_OUTPUT(rd));
endmodule
```

```
# test.pcf
set_io --warn-no-port PIN_5 C1
set_io --warn-no-port PIN_18 A9
```

```
$ yosys -p "synth_ice40 -top test; json -o test.json" test.v
$ nextpnr-ice40 --lp8k --package cm81 --pcf test.pcf --json test.json
...
Info: Packing constants..
Info: Packing IOs..
Info: PIN_5 feeds SB_IO in.lvds_in, removing $nextpnr_ibuf PIN_5.
terminate called after throwing an instance of 'nextpnr_ice40::assertion_failure'
  what():  Assertion failure: wire != WireId() (/build/source/ice40/arch.h:610)
```

smunaut, 2019-01-18:
https://github.com/YosysHQ/nextpnr/pull/213

This will show a slightly more understandable error message. (especially since the site id is printed and linked to the pin a couple lines earlier, so it should be clear enough).

For your case, if you want to still use this as a clock input, the only solution is to use a SB_IO followed by a separate SB_GB.  You loose the shorter path / deterministic timing of the SB_GB_IO but if you don't use that clock to capture any IO linked to it, that doesn't matter at all.

bgamari, 2019-01-19:
Ahh, silly me. Of course.

Thanks @smunaut!

closed 2019-01-19 by bgamari