YosysHQ/nextpnr #215: Boost problem? CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'

gojimmypi, 2019-01-18:
I've been trying to get nextpnr to compile and have encountered a variety of issues as noted in https://github.com/YosysHQ/nextpnr/issues/214 and https://github.com/YosysHQ/nextpnr/issues/129#issuecomment-455707572. I believe my latest error of `No version of Boost::Python 3.x could be found` is misleading and inaccurate. Is this really a `boost` problem?

Given the whole issue with cmake cache and all, I removed all instances of python3.x and all respective libraries, reinstalled fresh like this:

```
sudo apt-get install python3.5
sudo apt-get install qt5-default python3.5-dev
sudo apt-get install libboost-all-dev
```
Then I cloned nextpnr into a completely new directory in order to run `cmake -DARCH=ice40 .`  

The fresh output is:

```
0 $ cmake -DARCH=ice40 .
-- The C compiler identification is GNU 4.8.4
-- The CXX compiler identification is Clang 3.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3.5 (found suitable version "3.5.2", minimum required is "3.5")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "3.5")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE
-- Boost version: 1.54.0
-- Found the following Boost libraries:
--   filesystem
--   thread
--   program_options
--   system
--   chrono
--   date_time
--   atomic
-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libGL.so
CMake Error at CMakeLists.txt:170 (message):
  No version of Boost::Python 3.x could be found.

-- Configuring incomplete, errors occurred!
See also "/home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeOutput.log".
See also "/home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeError.log".

```
However when looking into the contents of` CMakeError.log` - I find the first error appears to be `CheckSymbolExists.c:(.text+0x16): undefined reference to pthread_create` as shown below:
```
Determining if the pthread_create exist failed with the following output:
Change Dir: /home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_90c9b/fast"
/usr/bin/make -f CMakeFiles/cmTC_90c9b.dir/build.make CMakeFiles/cmTC_90c9b.dir/build
make[1]: Entering directory `/home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_90c9b.dir/CheckSymbolExists.c.o
/usr/bin/cc     -o CMakeFiles/cmTC_90c9b.dir/CheckSymbolExists.c.o   -c /home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTm$Linking C executable cmTC_90c9b
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_90c9b.dir/link.txt --verbose=1
/usr/bin/cc       CMakeFiles/cmTC_90c9b.dir/CheckSymbolExists.c.o  -o cmTC_90c9b -rdynamic
CMakeFiles/cmTC_90c9b.dir/CheckSymbolExists.c.o: In function `main':
CheckSymbolExists.c:(.text+0x16): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [cmTC_90c9b] Error 1
make[1]: Leaving directory `/home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTmp'
make: *** [cmTC_90c9b/fast] Error 2

File /home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_99b0f/fast"
/usr/bin/make -f CMakeFiles/cmTC_99b0f.dir/build.make CMakeFiles/cmTC_99b0f.dir/build
make[1]: Entering directory `/home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_99b0f.dir/CheckFunctionExists.c.o
/usr/bin/cc    -DCHECK_FUNCTION_EXISTS=pthread_create   -o CMakeFiles/cmTC_99b0f.dir/CheckFunctionExists.c.o   -c /usr/$Linking C executable cmTC_99b0f
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_99b0f.dir/link.txt --verbose=1
/usr/bin/cc   -DCHECK_FUNCTION_EXISTS=pthread_create    CMakeFiles/cmTC_99b0f.dir/CheckFunctionExists.c.o  -o cmTC_99b0$/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
make[1]: *** [cmTC_99b0f] Error 1
make[1]: Leaving directory `/home/gojimmypi/temp/nextpnr/CMakeFiles/CMakeTmp'
make: *** [cmTC_99b0f/fast] Error 2

```

So is this really a `No version of Boost::Python 3.x could be found` problem or is something else wrong?

daveshah1, 2019-01-21:
The Boost::Python issue does appear to be a genuine problem - whereas the error in the CMake log looks like one of CMake's internal checks for functionality failing, which doesn't seem to be a wider problem - although it didn't find pthread_create in pthreads; it did find it in pthread.

Some poking around (https://packages.ubuntu.com/trusty/amd64/libboost-python1.54-dev/filelist) suggests that the default Boost::Python for Ubuntu 14.04 only includes support for Python up to 3.4 (and nextpnr is designed for 3.5+).

There are three options here:
 - Build nextpnr without Python and UI (add `-DBUILD_PYTHON=OFF -DBUILD_GUI=OFF` to the CMake command line)
 - Find a version of Boost::Python with Python 3.5 support for Ubuntu 14.04, or build it from source
 - Upgrade to a newer Ubuntu version

gojimmypi, 2019-01-24:
@daveshah1 thank you! yes - that was exactly the problem for `CMake`, but now there's a new problem with `make`. 

```
$ make --version
GNU Make 4.1
Built for x86_64-pc-linux-gnu
Copyright (C) 1988-2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

```
Manually updating Ubuntu (WSL) fixed the CMake/boost issue for me:

```
sudo apt-get install update-manager-core
sudo do-release-upgrade
sudo apt-get update
sudo apt-get dist-upgrade

# doing this wiped out CMake, so I had to reinstall:
wget http://www.cmake.org/files/v3.13/cmake-3.13.3.tar.gz
tar xf cmake-3.13.3.tar.gz
cd cmake-3.13.3
./configure
make
sudo make install
cmake --version
```
Here's new version:
```
0 $ cat /etc/os-release
NAME="Ubuntu"
VERSION="16.04.5 LTS (Xenial Xerus)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 16.04.5 LTS"
VERSION_ID="16.04"
HOME_URL="http://www.ubuntu.com/"
SUPPORT_URL="http://help.ubuntu.com/"
BUG_REPORT_URL="http://bugs.launchpad.net/ubuntu/"
VERSION_CODENAME=xenial
UBUNTU_CODENAME=xenial

```
bulding `nextpnr` now works with CMake!  (but...)

```
$ cmake -DARCH=ice40 .
-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is Clang 3.8.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found PythonInterp: /usr/bin/python3.5 (found suitable version "3.5.2", minimum required is "3.5")
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable version "3.5.2", minimum required is "3.5")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
[... 149 lines cut ...]
[ 45%] Building CXX object CMakeFiles/ice40_chipdb.dir/ice40/chipdbs/chipdb-5k.cc.o
[ 45%] Built target ice40_chipdb
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
I tried reinstalling boost, but things seemed happy:

```
$ sudo apt-get install qt5-default
[sudo] password for gojimmypi:
Reading package lists... Done
Building dependency tree
Reading state information... Done
qt5-default is already the newest version (5.5.1+dfsg-16ubuntu7.5).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.

0 $ sudo apt-get install libboost-all-dev
Reading package lists... Done
Building dependency tree
Reading state information... Done
libboost-all-dev is already the newest version (1.58.0.1ubuntu1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
```

As this new error is perhaps unrelated to boost, perhaps we close this issue and open a new one?

daveshah1, 2019-01-24:
This seems to be a manifestation of https://bugs.llvm.org/show_bug.cgi?id=22924

directysj, 2021-04-06:
I am building Faunus software using cmake with command -->  $cmake . -DENABLE_OPENMP=ON -DENABLE_PYTHON=ON -DCMAKE_INSTALL_PREFIX:PATH=/home/user/localInstall/tstFaunusInstOpenMP -DPYTHON_EXECUTABLE=/home/user/anaconda3/envs/faunus-env/bin/python3.6 -DPYTHON_INCLUDE_DIR=/home/user/anaconda3/envs/faunus-env/include/python3.6m -DPYTHON_LIBRARY=/home/user/anaconda3/envs/faunus-env/lib -DCMAKE_CXX_FLAGS_DEBUG="-pthread" 

I am getting build error like :

Determining if the pthread_create exist failed with the following output:
Change Dir: /home/user/softwares/openMPfaunus-2.4.1/CMakeFiles/CMakeTmp

Run Build Command:"/usr/bin/make" "cmTC_a43c8/fast"
/usr/bin/make -f CMakeFiles/cmTC_a43c8.dir/build.make CMakeFiles/cmTC_a43c8.dir/build
make[1]: Entering directory '/home/user/softwares/openMPfaunus-2.4.1/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_a43c8.dir/CheckSymbolExists.c.o
/usr/bin/cc    -o CMakeFiles/cmTC_a43c8.dir/CheckSymbolExists.c.o   -c /home/user/softwares/openMPfaunus-2.4.1/CMakeFiles/CMakeTmp/CheckSymbolExists.c
Linking C executable cmTC_a43c8
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_a43c8.dir/link.txt --verbose=1
/usr/bin/cc      CMakeFiles/cmTC_a43c8.dir/CheckSymbolExists.c.o  -o cmTC_a43c8 
/usr/bin/ld: CMakeFiles/cmTC_a43c8.dir/CheckSymbolExists.c.o: in function `main':
CheckSymbolExists.c:(.text+0x1b): undefined reference to `pthread_create'
collect2: error: ld returned 1 exit status
make[1]: *** [CMakeFiles/cmTC_a43c8.dir/build.make:87: cmTC_a43c8] Error 1
make[1]: Leaving directory '/home/user/softwares/openMPfaunus-2.4.1/CMakeFiles/CMakeTmp'
make: *** [Makefile:121: cmTC_a43c8/fast] Error 2

File /home/user/softwares/openMPfaunus-2.4.1/CMakeFiles/CMakeTmp/CheckSymbolExists.c:
/* */
#include <pthread.h>

int main(int argc, char** argv)
{
  (void)argv;
#ifndef pthread_create
  return ((int*)(&pthread_create))[argc];
#else
  (void)argc;
  return 0;
#endif
}

##############################################################
What I could understand from this is that there is section of file "CheckSymbolExists.c", which requires pthread_create. I wrote a simple code in C and ran using option: gcc example.c -lm -pthread and it's is working fine but if I run without -pthread option I am getting the same error concerning pthread_create. 
Can someone help out this.

gatecat, 2022-08-18:
Linked LLVM issue should now be fixed and hopefully all is working.

For any nextpnr uses finding this issue in the future, and want to avoid the hassle of building it altogether, prebuilt binaries for both Linux (WSL-compatible) and native Windows are available as part of https://github.com/YosysHQ/oss-cad-suite-build/releases

closed 2022-08-18 by gatecat