YosysHQ/nextpnr #218: nextpnr max freq unchanged by seeds (though icetime shows changes)

jlwehle, 2019-01-24:
Running the try-seeds script results in icetime showing total path
delays corresponding to anywhere from 202 MHz to 281 MHz.

The odd thing is that nextpnr-ice40 shows the maximum clock frequency
in all forty cases as 154 MHz.

I understand that the two programs may use different timing modules ...
it just seems suspicions that nextpnr-ice40 doesn't show different
values for different seeds which clearly are producing different
results.

BTW: What are the thoughts as far as to which program tends to be more
realistic with regards to the maximum frequency?

-- John

jlwehle, 2019-01-24:
Here's the testcase.

[testcase.zip](https://github.com/YosysHQ/nextpnr/files/2790195/testcase.zip)

jlwehle, 2019-01-24:
Sorry ... hit the wrong button :-)

daveshah1, 2019-01-24:
The reason for the lack of changes in nextpnr is to do with how nextpnr considers paths.

icetime considers paths between IO and all registers when counting Fmax, whereas nextpnr only considers interior paths (between registers) and reports the register to/from IO paths separately as max delay to/from '$async'.

In this case, the critical path appears to be between two registers in adjacent tiles. As a result, there is little to change when changing seeds (whereas the routing to/from IO does change which affects icetime's Fmax and the separate delays that nextpnr prints). 

Note that nextpnr prints a lower Fmax than icetime because nextpnr correctly handles paths between the positive and negative edge of a clock, unlike icetime.

jlwehle, 2019-01-25:
Thanks for the explanation.

In looking at the ICE40 Family Handbook I saw information regarding the timing
of typical building blocks and external timing, however I didn't see information
regarding the internal routing delays (i.e. adjacent versus span 4 versus span 12).

Did I miss it or is there a technical note that gives more details?