YosysHQ/nextpnr #224: timing analysis fails using Jan 29, 2019 code.

jlwehle, 2019-02-05:
Running nextpnr from Jan 29, 2019 like so:

  nextpnr-ice40 --hx8k --package tq144:4k --asc rx_cdr_bbpd.asc --json rx_cdr_bbpd.json --freq 350 --seed 4

produces:

  ERROR: timing analysis failed due to presence of combinatorial loops, incomplete specification of timing ports, etc.

I believe this worked in the past using nextpnr from Dec 19, 2018.

The file is in:
[testcase.zip](https://github.com/YosysHQ/nextpnr/files/2830746/testcase.zip)

jlwehle, 2019-02-05:
That's strange ... looking at the verilog source being processed by yosys like so:

  yosys -q -p 'synth_ice40 -top rx_cdr_bbpd -json rx_cdr_bbpd.json' rx_cdr_bbpd.v

I'm failing to see the combinational loop.  The output design is
for a bang bang phase detector.  The combinational output is determined
soley by the state of the flip flops.  The flip flops are in series
and ultimately feed from the input.

[bbpdv.zip](https://github.com/YosysHQ/nextpnr/files/2833206/bbpdv.zip)

daveshah1, 2019-02-05:
The problem is
```verilog
  always @*
    case (s[3:1])
      3'b001: late = 1'b1;
      3'b100: early = 1'b1;
      default: begin
                 early = 1'b0;
                 late = 1'b0;
               end
    endcase
```
When `s[3:1]` is 3'b001 or 3'b100, only either `late` or `early` is set, the other one remaining at its previous value. This will result in an inferred latch. As the iCE40 doesn't have latch primitives, Yosys has to implement this as a combinational loop.

jlwehle, 2019-02-06:
Now that you point it out I see it exactly.
Thanks ... I appreciate you taking the time to point out a classic
mistake.

closed 2019-02-06 by jlwehle