YosysHQ/nextpnr #229: ERROR: Cell '$gbuf_SB_GB_IO_io' cannot be bound to bel 'X0/Y17/gb' since it is already bound to cell '$gbuf_SB_PLL40_CORE_pllout_b'

whitequark, 2019-02-09:
To reproduce, use this archive: [vga-output.zip](https://github.com/YosysHQ/nextpnr/files/2847603/vga-output.zip)

I don't entirely understand the nature of the bug. I think it's related to global promotion. The error message refers to (a) a bel that has never been mentioned in the log before, and (b) an internal name of a global buffer that has never been mentioned in the log before, so at least the error message is wrong. 

But, I also don't know how to make the design work; switching to PLLOUTGLOBAL instead of PLLOUTCORE has no effect.

daveshah1, 2019-02-09:
Thanks for the report and testcase!

#230 fixes the problem, if possible it would be much appreciated if you could confirm this works on hardware.

BTW there is no `PLLOUTGB` port so the testcase as is doesn't synthesise, but changing it to either `PLLOUTCORE` or `PLLOUTGLOBAL` works now (and didn't before). cc @eddiehung if you are adding this to the regressions.

whitequark, 2019-02-09:
> BTW there is no `PLLOUTGB` port so the testcase as is doesn't synthesise, but changing it to either `PLLOUTCORE` or `PLLOUTGLOBAL` works now (and didn't before).

I archived the wrong file, sorry. This was a typo.

whitequark, 2019-02-09:
@daveshah1 BTW, can you confirm what happens with the SB_IO that is eaten by the PLL? Is it correctly placed into high-Z statea? Is the pullup enabled?

daveshah1, 2019-02-09:
It will be high-Z (unless an output is used at that location), and the pullup enabled as is the default unused IO (placing an IO at this location with the input unused and the pullup disabled should override this if needed, but I haven't tested)

closed 2019-02-09 by daveshah1

eddiehung, 2019-02-09:
Testcase added to https://github.com/YosysHQ/nextpnr-tests/commit/d3d81370e33cab1e2a0c74f0a334d17c80a0ee18

PR #230 (2019-02-09): ice40: Don't create PLLOUT_B buffer for single-output PLL variants
Fixes #229 

For single-output PLL variants, we were only checking for conflicts with the first output; but still creating SB_GBs for both outputs.