YosysHQ/nextpnr #233: segmentation fault while placing rv64imadc based blinky

gsomlo, 2019-02-10:
When I updated from (working) commit db990d2 to commit a520b38, my p&r run generating .config out of the yosys-produced .json started failing with a segmentation fault:
```
...
Info:   at iteration #575: temp = 0.000001, cost = 217601, est tns = -9702.30ns
Info:   at iteration #580: temp = 0.000000, cost = 217597, est tns = -9702.54ns
Info:   at iteration #581: temp = 0.000000, cost = 217597.000000
Info: SA placement time 11177.53s

Info: Max frequency for clock '$glbnet$clock': 10.09 MHz (PASS at 10.00 MHz)

Info: Max delay <async>               -> posedge $glbnet$clock: 6.95 ns
Info: Max delay posedge $glbnet$clock -> <async>              : 16.12 ns

/usr/include/c++/8/bits/stl_vector.h:932: std::vector<_Tp, _Alloc>::reference   std::vector<_Tp, _Alloc>::operator[](std::vector<_Tp, _Alloc>::size_type) [with _Tp = unsigned int; _Alloc = std::allocator<unsigned int>; std::vector<_Tp,     _Alloc>::reference = unsigned int&; std::vector<_Tp, _Alloc>::size_type = long  unsigned int]: Assertion '__builtin_expect(__n < this->size(), true)' failed.
make: *** [Makefile:102: top.config] Aborted (core dumped)
```
I didn't have the debugging sources installed, so all I got from `coredumpctl` was this:
```
           PID: 32201 (nextpnr-ecp5)
        Signal: 6 (ABRT)
     Timestamp: Sat 2019-02-09 21:33:13 EST (9h ago)
  Command Line: nextpnr-ecp5 --json top.json --lpf versa.lpf --basecfg /usr/share/prjtrellis/misc/basecfgs/empty_lfe5um-45f.config --um-45k --freq 10 --textcfg top.config
    Executable: /usr/bin/nextpnr-ecp5
       Message: Process 32201 (nextpnr-ecp5) of user 1000 dumped core.
                
                Stack trace of thread 32201:
                #0  0x00007f204f52deab __GI_raise (libc.so.6)
                #1  0x00007f204f5185b9 __GI_abort (libc.so.6)
                #2  0x0000558eca99bd48 _ZSt20__replacement_assertPKciS0_S0_ (nextpnr-ecp5)
                #3  0x0000558eca9fd034 _ZN12nextpnr_ecp515timing_analysisEPNS_7ContextEbbbb (nextpnr-ecp5)
                #4  0x0000558eca9bba00 _ZN12nextpnr_ecp58SAPlacer5placeEv (nextpnr-ecp5)
                #5  0x0000558eca9b4729 _ZN12nextpnr_ecp57placer1EPNS_7ContextENS_10Placer1CfgE (nextpnr-ecp5)
                #6  0x0000558ecaa24148 _ZN12nextpnr_ecp54Arch5placeEv (nextpnr-ecp5)
                #7  0x0000558eca9990af _ZN12nextpnr_ecp514CommandHandler11executeMainESt10unique_ptrINS_7ContextESt14default_deleteIS2_EE (nextpnr-ecp5)
                #8  0x0000558eca9994c0 _ZN12nextpnr_ecp514CommandHandler4execEv (nextpnr-ecp5)
                #9  0x0000558eca992f98 main (nextpnr-ecp5)
                #10 0x00007f204f51a11b __libc_start_main (libc.so.6)
                #11 0x0000558eca9938ea _start (nextpnr-ecp5)
```
I'm still using prjtrellis commit 5b6bc70 in case the couple of newer commits on top of that are somehow related.

Applying PR #219 on top of the current master (commit 6cd0b8a) works fine, and I haven't yet tried 6cd0b8a without the #219 patches, nor have I tried the latest master of prjtrellis.

I can run more detailed tests and report back, but without #219 it takes about 4 hours (into the roughly 5-hour total P&R run time) before I get a failure I can investigate :)

eddiehung, 2019-02-11:
Backtrace seems to allude to a timing analysis error.

Yesterday I merged PR #226 which could be what you're seeing too.
Sorry to hear that it only fails after 4+ hours!

gsomlo, 2019-02-11:
I ran the whole thing again with commit 6cd0b8a (build using the latest available prjtrellis as well). This time I had loaded the debuginfo symbols as well, and attached is the output of gdb's backtrace: 
[nextpnr-core.txt](https://github.com/YosysHQ/nextpnr/files/2849702/nextpnr-core.txt)

Hope this helps narrow it down a bit -- Thanks!

eddiehung, 2019-02-11:
Hi @gsomlo, from the backtrace it looks like you are using commit 6cd0b8a which is after PR #226 so that didn't fix it for you.

The failing line: https://github.com/YosysHQ/nextpnr/blob/6cd0b8a12345a635351122ed3e4e635162767b3a/common/timing.cc#L937
is incidentally the one that #226 was meant to fix. 

Let me see if I can reproduce using the 4 hour test case. Standby!

eddiehung, 2019-02-11:
Looks like you are using ecp5 which I haven't got set up yet, so this might take a little while.

In the mean time, I did some back of the envelope calculations, and it is possible you are so (un)lucky to have stumbled across an off-by-one, so can you try PR #234?

Thanks!

gsomlo, 2019-02-11:
On Mon, Feb 11, 2019 at 04:34:50PM +0000, Eddie Hung wrote:
> Looks like you are using ecp5 which I haven't got set up yet, so this might
> take a little while.
> 
> In the mean time, I did some back of the envelope calculations, and it is
> possible you are so (un)lucky to have stumbled across an off-by-one, so can
> you try PR #234?

Building nextpnr with PR #234 right now, will fire off the
yoloRISC/rocket/ecp5 build next, then report back in 4-5 hours :)
Thanks!

gsomlo, 2019-02-12:
With PR #234, p&r completes successfully, with no more segfaults.
I tested the resulting bitstream on the versa board, works fine
as well.

Thanks again,
--Gabriel

eddiehung, 2019-02-12:
Smegcellent! Closing with the merge of #234.

closed 2019-02-12 by eddiehung