YosysHQ/nextpnr #235: CR LF Problems when processing lpf files

gojimmypi, 2019-02-16:
When nextpnr processes an lpf file such as [this one](https://github.com/gojimmypi/ULX3S-Blinky/blob/development/ulx3s_v20.lpf), errors are encountered on some elements that are otherwise not problematic if there are only LF delimiters between lines.

For instance, note the CR LF chars here in this source file that causes errors:

[image]

This same file, with only LF delimiters, does not cause the error:
[image]
 
Here's a comparison of the same processing of nextpnr, but the one with the extra CR char causes errors:

[image]
[..snip..]
[image]

The shocking thing there is I discovered this issue simply when doing a `git clone` of a project in a Windows command prompt. It seems that git decided to take things upon itself to quietly replace all the LF chars with CRLF at clone time. So ya, perhaps this is also a git issue, as well.

See also the [diff of lpf files](https://github.com/gojimmypi/ULX3S-Blinky/commit/0d5dc3ee681d280afd2fe19d3462c417378e690a#diff-338eb33496f9c34e1d339675d17d8477) as well as the [diff of output log files](https://github.com/gojimmypi/ULX3S-Blinky/commit/ad5a4e4a464cc997dd6c5d4c99f89840113676bc#diff-ffa9d213659f6ecce35617dee1fc8517L1617) in my fork.

gojimmypi, 2019-02-25:
I've created a project to demonstrate this error:

https://github.com/gojimmypi/nextpnr-issue235

daveshah1, 2019-02-27:
8744c46 (part of #243) should fix this and also correctly handle uppercase frequencies.

gojimmypi, 2019-02-28:
although it appears the CR/LF ending blank line problem is resolved (as well as many of the warnings), I'm now seeing a new error with uppercase frequency that I did not previously see.

```
Warning:     ignoring unsupported LPF command 'BLOCK RESETPATHS'
Warning:     ignoring unsupported LPF command 'BLOCK ASYNCPATHS'
ERROR: unsupported frequency unit 'MHZ'
```

pulled and installed fresh `prjtrellis` as well as `nextpnr`.

`nextpnr-ecp5 -- Next Generation Place and Route (git sha1 a2d906a)`

```
gojimmypi:~/workspace/prjtrellis
$ git pull
Updating 69b48a5..9216745
Fast-forward
 database                            |  2 +-
 examples/versa5g/make_14seg.py      | 22 +++++++++++++++++-----
 fuzzers/ECP5/144-bootaddr/fuzzer.py | 45 +++++++++++++++++++++++++++++++++++++++++++++
 libtrellis/include/Bels.hpp         |  2 +-
 libtrellis/include/Bitstream.hpp    |  5 ++++-
 libtrellis/src/Bels.cpp             | 13 +++++++++----
 libtrellis/src/Bitstream.cpp        | 39 ++++++++++++++++++++++++++++++++++++++-
 libtrellis/src/Chip.cpp             | 28 ++++++++++++++++++----------
 libtrellis/src/RoutingGraph.cpp     |  2 --
 libtrellis/tools/ecppack.cpp        |  4 ++++
 libtrellis/tools/ecpunpack.cpp      | 16 +++++++++++++++-
 11 files changed, 152 insertions(+), 26 deletions(-)
 create mode 100644 fuzzers/ECP5/144-bootaddr/fuzzer.py
```

```
gojimmypi:~/workspace/nextpnr
$ git pull
Updating 4c73061..a2d906a
Fast-forward
 .cirrus/Dockerfile.ubuntu16.04 |   4 +-
 common/design_utils.cc         |   4 +
 common/timing.cc               |   9 +-
 ecp5/arch.cc                   | 161 +++++++++++++---
 ecp5/arch.h                    |  10 +-
 ecp5/arch_place.cc             |  74 +++++++
 ecp5/archdefs.h                |   2 +
 ecp5/bitstream.cc              | 144 +++++++++++++-
 ecp5/cells.cc                  |   3 +
 ecp5/cells.h                   |  11 ++
 ecp5/constids.inc              |   2 +
 ecp5/docs/primitives.md        |  47 +++++
 ecp5/globals.cc                |  19 +-
 ecp5/lpf.cc                    |  25 ++-
 ecp5/pack.cc                   | 854 ++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++-
 ecp5/trellis_import.py         |  27 ++-
 ice40/arch.cc                  |  16 +-
 ice40/arch.h                   |   5 +-
 ice40/arch_pybindings.cc       |   1 +
 ice40/bitstream.cc             |  12 +-
 ice40/chipdb.py                |  23 ++-
 ice40/constids.inc             |   6 +
 ice40/delay.cc                 |   2 +-
 ice40/family.cmake             |  16 +-
 ice40/main.cc                  |   8 +-
 ice40/project.cc               |   3 +
 ice40/resource/chipdb.rc       |   1 +
 ice40/resource/embed.cc        |   4 +-
 ice40/resource/resource.h      |   1 +
 tests                          |   2 +-
 30 files changed, 1426 insertions(+), 70 deletions(-)
 create mode 100644 ecp5/docs/primitives.md
```

daveshah1, 2019-02-28:
I'm slightly unsure what you mean in your edit; but at least for me all the problems are resolved in the ecp5lpf branch.

gojimmypi, 2019-02-28:
when I originally posted my comment, I didn't realize the commit for this fix was to a different branch. The master branch still had the mixed case MHz compare. 

yes; confirmed working well in the `ecp5lpf` branch for my demo script, above.

I now see only one warning for `SYSCONFIG`, and no errors for both unix and dos format lpf files.

`Warning:     ignoring unsupported LPF command 'SYSCONFIG CONFIG_IOVOLTAGE=3.3 COMPRESS_CONFIG=ON MCCLK_FREQ=62 MASTER_SPI_PORT=DISABLE SLAVE_SPI_PORT=DISABLE SLAVE_PARALLEL_PORT=DISABLE' (on line 14)`

daveshah1, 2019-02-28:
Created regression test for this in YosysHQ/nextpnr-tests#1

closed 2019-02-28 by daveshah1

ecp5fpgauser, 2020-05-12:
Even if this ticket is closed, I am afraid I face the same issue, ie. "ignoring unsupported LPF command 'BLOCK_JTAGPATHS'" and the same warning message for 'SYSCONFIG MASTER_SPI_PORT=ENABLE TRANSFR=OFF SLAVE_SPI_PORT=DISABLE CONFIG_IOVOLTAGE=3.3'. Is there an option to work around this issue ?

smunaut, 2020-05-12:
They are just warning right ? So it shouldn't prevent it from continuing.

ecp5fpgauser, 2020-05-12:
Yes, that's right, I can continue the flow, but at the end, no LED is blinking on this ECP5 Evaluation Board Default Demo provided by Lattice here: http://www.latticesemi.com/products/developmentboardsandkits/ecp5evaluationboard#_C694C444BC684AD48A3ED64C227B6455. Target is LFE5UM5G-45F on VERSA dev kit. As I face no other error/warning except the one I have already spoken, I try to find where something goes wrong.

daveshah1, 2020-05-12:
Have you modified the pins for the ECP5 evaluation board? That warning should definitely not cause this failure. You could also try the ecp5_evn example in trellis to make sure your environment is set up correctly.

ecp5fpgauser, 2020-05-12:
I checked the pins in LPF file and it seems it was a setting for another target! Now everything works fine, including both ECP5 eval board demo and trellis/versa5g demo, even if I still face warning previously written ""ignoring unsupported LPF command SYSCONFIG...". Thanks very much for your help!