YosysHQ/nextpnr #237: Clock driving logic causes timing analysis failure

osresearch, 2019-02-20:
Trying to output the negated version of the high frequency oscillator on the iCE40up5k results in a nextpnr timing error.  This code:

```
module top(
        output P1A1
);
        wire clk;
        SB_HFOSC osc(
                .CLKHFPU(1'b1),
                .CLKHFEN(1'b1),
                .CLKHF(clk)
        );

        assign P1A1 = !clk;

endmodule
```

results in

```
yosys \
	-q \
	-p 'read_verilog cycle.v' \
	-p 'synth_ice40 -top top -json cycle.json' \
	-E .cycle.d \

nextpnr-ice40 \
	-q \
	--up5k \
	--package sg48 \
	--asc cycle.asc \
	--pcf icebreaker.pcf \
	--json cycle.json \

ERROR: timing analysis failed due to presence of combinatorial loops, incomplete specification of timing ports, etc.
0 warnings, 1 error
```

 The PCF contains:
```
set_io -nowarn P1A1 4
```

This is with `Yosys 0.7+631 (git sha1 93efbd5, clang  -fPIC -Os)` and `nextpnr-ice40 -- Next Generation Place and Route (git sha1 eb456ef)`.

smunaut, 2019-02-20:
Obviously a bug, but as a side note, that's really not how you should be outputting a clock to a pin if you can avoid it.

daveshah1, 2019-02-20:
#238 should fix this

PR #238 (2019-02-20): ice40: Fix timing class of 'padin' GB outputs
For the `padin` case, the global buffer input is not connected so treating as combinational is not appropriate.

Fixes #237
closed 2019-02-20 by daveshah1