YosysHQ/nextpnr #240: nextpnr --gui qt5.glx error

ric96, 2019-02-22:
While running nextpnr-ice40 --gui I am getting qt5.glx error
```
[ric@manjaro-c2q nextpnr]$ nextpnr-ice40 --gui
qt5ct: using qt5ct plugin
qt5ct: D-Bus global menu: no
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (8 8 8 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 8 8 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 8 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (8 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (8 8 8 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 8 8 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 8 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 0)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (8 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 8 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 8 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
qt.glx: qglx_findConfig: Failed to finding matching FBConfig (1 1 1 8)
[ric@manjaro-c2q nextpnr]$ 

```

I can only see this on Arch Linux and Manjaro Linux, but apps like qtbrowser seem to be working fine.
```
[ric@manjaro-c2q nextpnr]$ qmake-qt5 --version
QMake version 3.1
Using Qt version 5.12.1 in /usr/lib
[ric@manjaro-c2q nextpnr]$ 

```

ric96, 2019-02-22:
Attaching strace
[nextpnrgui_strace.txt](https://github.com/YosysHQ/nextpnr/files/2895083/nextpnrgui_strace.txt)

daveshah1, 2019-02-25:
To help track this down: what GPU and driver are you using?

ric96, 2019-02-25:
Radeon 6450 with Mesa 18.3.3, I have also tried Nvidia gt710 + Mesa with the same result
Attaching glxinfo.
Also I can see the window, but every-time the window in in the foreground the screen goes blank as long as the window is behind another window I can see it fine.

[glxinfo.txt](https://github.com/YosysHQ/nextpnr/files/2899937/glxinfo.txt)

itviewer, 2019-04-11:
same problem with Qt 5.12.1, had changed to Qt version 5.9.5

sarthakpati, 2019-05-20:
I ran into this issue as well. [This post](https://stackoverflow.com/a/55235823/1228757) on SO might help.

ric96, 2019-05-22:
> I ran into this issue as well. [This post](https://stackoverflow.com/a/55235823/1228757) on SO might help.

that fixed it, submitting pr

closed 2022-08-18 by gatecat