YosysHQ/nextpnr #250: nextpnr default output is strerr and not stdout?

gojimmypi, 2019-03-15:
I'm using `nextpnr-ecp5 -- Next Generation Place and Route (git sha1 cadbf42)`

I noticed when writing a script to send `nextpnr` output to a log for later review... that the log file is _empty_:

```
	nextpnr-ecp5 --25k --json $(PROJ).json  \
		--lpf ulx3s_v20.lpf \
		--textcfg $(PROJ).out.config | tee -a  $(PROJ).nextpnr-ecp5.log
```

I think this explains it:

```
nextpnr-ecp5 --help > myout.log 2> myerr.log
ls my*.log -al
```

The file `myerr.log` has the help output, and `myout.log` is empty. 

I double-checked [here](https://askubuntu.com/questions/625224/how-to-redirect-stderr-to-a-file) to confirm the first `>` is `stdout` and `2>` is `stderr`.

Or this example:

`nextpnr-ecp5 --help > myout.log`  (output seen on screen, should be only in `myout.log`)

`nextpnr-ecp5 --help 2> myout.log` (no output seen on screen, but was written to file)

Is this the intended behavior?

daveshah1, 2019-03-15:
imo it is more idiomatic for status/progress information (ie all of nextpnr's log output) to be written to stderr - although we don't support this at the moment, theoretically stdout could be used for output data (ie instead of writing a bitstream to a file).

In any case you can use --log to write a log file rather than needing to use tee.

closed 2019-03-15 by daveshah1

gojimmypi, 2019-03-15:
an interesting design choice. I probably don't agree that successful progress/status information should be written to `stderr` as a default. It would be nice to have the capability to have errors logged separately; that's the whole point of `stderr`. (although `grep` could be used as a brute-force, incomplete, not-fully-reliable solution to find errors, and perhaps warnings)

but I do see your point on one day piping the bitstream output in some future version. In that case though, the `--quiet` could be used, along with `--log` as desired, and a new option such as `--pipe-bitstream` that only sends the bitstream to `stdout`, no?  (in this case, errors and progress would all go to the log file)

just my opinion / idea. :)  thanks for the prompt reply. cheers

gojimmypi, 2019-03-16:
one additional observation:  `yosys` sends all output to `stdout` unlike `nextpnr` that sends all output to `stderr` as noted above. 

perhaps it would be best if they were consistent? one could argue the same output of `yosys` might be streamed (to a file), the way `nextpnr` would stream a bitfile, no?