YosysHQ/nextpnr #253: Missing parameter assignment results in Warning: No clocks found in design

gojimmypi, 2019-03-21:
I have a fork of the [DoctorWkt/ULX3S-Blinky](https://github.com/DoctorWkt/ULX3S-Blinky) located [here](https://github.com/gojimmypi/ULX3S-Blinky.git).

If I clone and build it it like this:

```
git clone https://github.com/gojimmypi/ULX3S-Blinky.git blinky-test
cd blinky-test
make -f Makefile.12F ulx3s.bit
```
It builds successfully with just one warning:

```
Info: constraining clock net 'clk_25mhz' to 25.00 MHz
Warning:     ignoring unsupported LPF command 'SYSCONFIG CONFIG_IOVOLTAGE=3.3 COMPRESS_CONFIG=ON MCCLK_FREQ=62 MASTER_SPI_PORT=ENABLE SLAVE_SPI_PORT=DISABLE SLAVE_PARALLEL_PORT=DISABLE' (on line 12)
```

The original `always` block looks like this:

```
    always @(posedge i_clk) begin
          ctr <= ctr + 1;
          o_led[7] <= 1;
          o_led[6] <= btn[1];
          o_led[5:0] <= ctr[23:18];
    end
```

However, when editing the `always` block, to remove one of the `o_led` assignments:

```
    always @(posedge i_clk) begin
          ctr <= ctr + 1;
          o_led[7] <= 1;
          o_led[6] <= btn[1];
          // o_led[5:0] <= ctr[23:18];
    end
```
I receive _3 warnings_:

```
Info: constraining clock net 'clk_25mhz' to 25.00 MHz
Warning:     ignoring unsupported LPF command 'SYSCONFIG CONFIG_IOVOLTAGE=3.3 COMPRESS_CONFIG=ON MCCLK_FREQ=62 MASTER_SPI_PORT=ENABLE SLAVE_SPI_PORT=DISABLE SLAVE_PARALLEL_PORT=DISABLE' (on line 12)

[...snip ...]

Info: SA placement time 0.04s
Warning: No clocks found in design

Info: Clock 'clk_25mhz' has no interior paths

Info: Max delay posedge clk_25mhz -> <async>: 3.64 ns

Info: Slack histogram:

[... snip ...]

Info: Route time 0.01s
Info: Checksum: 0x820bd924
Warning: No clocks found in design

Info: Critical path report for cross-domain path 'posedge clk_25mhz' -> '<async>':
Info: curr total
Info:  0.5  0.5  Source $auto$simplemap.cc:420:simplemap_dff$149_SLICE.Q0
Info:  1.5  2.1    Net led[6] budget 82.807999 ns (2,26) -> (0,17)
Info:                Sink led[6]$tr_io.I
Info: 0.5 ns logic, 1.5 ns routing

Info: Clock 'clk_25mhz' has no interior paths

Info: Max delay posedge clk_25mhz -> <async>: 2.06 ns

```

This may be helpful in duplicating the result and viewing only the warnings:

```
nextpnr-ecp5 --25k --json blinky.json  \
        --lpf ulx3s_v20.lpf \
        --textcfg ulx3s_out.config |& grep -i "warning"
```

On one hand, having a warning that some of the output parameters are unused/unassigned is definitely a good thing. However - a message about `Warning: No clocks found in design` is not at all helpful nor intuitive. Admittedly I am an FPGA newbie  - but it took me a ridiculous amount of time to find this, mostly chasing _clocks_ in my other actual project. Any chance this error can be made more explicit? Thanks :)

btw - in the project where I originally discovered this, all of my output variables *are* assigned, but not all in the `always` block. I have two output clocks to A/D and D/A converters that are simply assigned in the module like this:

```
module ulx3s_adda
  (input  i_clk, 
...
	 output o_AD_CLK,
	 output o_DA_CLK,
);
...
	// A/D Input Clock
	// "The pipelined architecture of the AD9280 operates on both rising and falling edges of the input clock.
	// The AD9280 is designed to support a conversion rate of 32 MSPS; running the part at slightly faster clock rates may
	// be possible, although at reduced performance levels." (see AD9280 datasheet, page 15)
	wire o_AD_CLK;
	assign J2_AD_CLK = o_AD_CLK; // = i_clk;
	assign o_AD_CLK = clk_25mhz;

	// D/A Outout Clock
	// AD9708 TxDAC: 125 MSPS Update Rate
	// "The DAC output is updated following the rising edge of the clock as shown in Figure 1 and is designed to support a
	// clock rate as high as 125 MSPS."
	// output propagation delay tPD is typically 1ns (see AD9708 datasheet page 3)
	wire o_DA_CLK;
	assign J2_DA_CLK = o_DA_CLK;
	assign o_DA_CLK = clk_25mhz;
```

There are of course timing issues that I will be working on, but for a crude first pass - it does actually work! :)

Any suggestions on if / how I should do this to avoid the warnings will be appreciated. Thanks

daveshah1, 2019-03-21:
This warning is generated when there are no interior paths (ie paths between register output and input) for any clock in the design. This means it is unable to provide any maximum frequency figures. In fact this warning was added when someone was wondering why nextpnr wasn't producing an Fmax - normally this warning is a result of a design being optimised away excessively.

daveshah1, 2019-03-21:
I'm open to suggestions as to how it could be better worded though.

gojimmypi, 2019-03-21:
Yes, if this was actually a _clock_ issue, I would agree... however because the `o_led[5:0] <= ctr[23:18];` assignment was missing from the `always` block - it would have been fantastic if the error message named the specific register or wire and was something like:

`Warning: no assignment for oled parameter in module top`

And further, this is actually seems outright wrong and misleading when the `oled` parameter is unassigned:

`Info: Clock 'clk_25mhz' has no interior paths`

As indeed the clock in `assign i_clk= clk_25mhz;` is used in the `always @(posedge i_clk) begin`; that `always` block would be the interior path, no?

koriakin, 2019-03-21:
Note that nextpnr doesn't see your original design -- by the time yosys is done with your design, there really is no register left on the clock, and this is what nextpnr sees.  It also cannot see that you don't assign to oled, because yosys simplifies that to a constant driver using the initial value (which is a perfectly reasonable design as far as nextpnr is concerned).  If you would like a better warning about that, it needs to be in yosys.

gojimmypi, 2019-03-27:
@koriakin thanks for the information and suggestion. I did open https://github.com/YosysHQ/yosys/issues/898 however it was closed as being  "not actionable", and upon the explanation from @cliffordwolf I agree - however I still think the `Warning: No clocks found in design` from `nextpnr` leaves a bit to be desired.

@daveshah1  What do you think of adding something to the message such as `Warning: No clocks found in design; check for warnings in yosys` ?

closed 2020-07-05 by daveshah1